strands-project/strands_executive #126: Planning when edges appear blocked

hawesie, 2015-02-12:
If we observe that an edge is blocked when navigating, are we able to ensure we don't try to immediately traverse it again?

bfalacerda, 2015-02-12:
no way to do it right now. I'll think about it next week, any ideas are welcome

hawesie, 2015-02-12:
Temporarily remove the edge from the navigation mdp, or increase its cost to something high enough to force it along another route?

Jailander, 2015-02-12:
I think it could make sense to have different recovery behaviours depending on the edges we are crossing, this way it would be easier to replan in case of blockage on specific places.
Also to solve this the only solution that comes to mind is that we could have specific action servers that check for blockages on some edges and if its blocked return failure.