strands-project/strands_executive #129: Release of mdp_plan_executor

hawesie, 2015-02-12:
If we plan to use the mdp_plan_executor for the deployments (which I hope we will), we need to get it released. @bfalacerda what issues are holding this back?

bfalacerda, 2015-02-12:
I need the new definitions of the topological map merged

bfalacerda, 2015-02-18:
should i create a new package from scratch in hydro-release and start building up from there and then make a PR, or start from hydro-devel, and then try to merge both branches? I don't know if hydro-devel is too out-of-date to do it, so I'm more inclined for the first option

@hawsie, suggestions?

hawesie, 2015-02-18:
I suggest starting from hydro-release. hydro-devel is out of date. Once you've got the mdp executor sorted we can scrap the old branch (perhaps renaming hydro-release to indigo-devel)

hawesie, 2015-03-09:
Progress on this is needed soon please!

bfalacerda, 2015-03-09:
the code is almost there, and then we need to test this, along with https://github.com/strands-project/strands_navigation/issues/181#issuecomment-77851495.

I'm putting this on top of my list of priorities

hawesie, 2015-03-27:
Released.

closed 2015-03-27 by hawesie