strands-project/strands_executive #130: create an iCal-based routine generator

marc-hanheide, 2015-02-23:
using the `icalendar` package we can easily parse iCal format, e.g. from Google Calendar. This is parsed to generate a routine, e.g. looking at the `RRULE` to determine recurring events. The idea is that the DESCRIPTION field in iCal is used to program a routine by simply putting the action in to be scheduled. Question is how duration, priorities etc. can be encoded. To be discussed. @hawesie

marc-hanheide, 2015-03-02:
This is being developed in https://github.com/strands-project/strands_executive/tree/ical It shall use [`DailyRoutineRunner`](https://github.com/strands-project/strands_executive/blob/ical/task_executor/src/task_executor/task_routine.py#L164) to control the submission of tasks taken from he iCal source. 
@hawesie's idea was to have a yaml config that defines all the (default) parameters needed for a task, so that in iCal one only needs to give a specific key word (potentially with auto_correction) to indicate the task to minimise usage error. We could still overwrite parameters in the iCal description field by putting yaml there.

marc-hanheide, 2015-03-13:
Plan has changed a bit, implementation now based on JSON rendering of gCal and having gCal resolve the recurring events itself, by using the correct API calls, here's the right URL: `https://www.googleapis.com/calendar/v3/calendars/henry.strands%40hanheide.net/events?key=API_KEY_HERE&singleEvents=true&orderBy=startTime&maxResults=2500` (`singleEvents=true` resolves recurrences)
Also, I'll not use `DailyRoutineRunner`, but rather a much simpler Runner.

hawesie, 2015-03-13:
Sounds sensible.

hawesie, 2015-03-13:
I think we should aim for some rostests that use both a daily routine and ical in parallel and test that they work out.

marc-hanheide, 2015-03-13:
Is that necessary? Right now I see them as independent. Actually, why would one still use the other routine in parallel? Maybe I miss something. But certainly running both in parallel shouldn't be a problem. Just the scheduler would see a few more tasks as they a submitted by both components independently. Right?

marc-hanheide, 2015-03-13:
I should add that all the recurrence management now works properly so it's easy to set things like every weekday mo-fr at 9 etc...

marc-hanheide, 2015-03-13:
Does it make any sense for the scheduler to see tasks way in advance before their requested start time? I mean practically... And if, what is a good time? Also, can I delete tasks from the scheduler I have once sent?

hawesie, 2015-03-13:
They need to be send before the execution time if there is likely to be anything else going on in the same time window. I think the daily routine sends stuff an hour in advance. 

Regarding having the routine too, you need that for charging, overnight tasks etc. but they should all play nicely together.

marc-hanheide, 2015-03-16:
I consider this kind of working until I hear (or experience) otherwise. It's implemented just will probably need bugfixing.

closed 2015-03-16 by marc-hanheide