strands-project/strands_executive #138: strands_executive_msgs missing in in `build_depend`

marc-hanheide, 2015-03-23:
see https://lcas.lincoln.ac.uk/jenkins/job/ros-indigo-wait-action_binarydeb_trusty_amd64/1/console

```
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by
  "strands_executive_msgs" with any of the following names:

    strands_executive_msgsConfig.cmake
    strands_executive_msgs-config.cmake

```

hawesie, 2015-03-23:
Nothing has changed in any of these packages! Could it be a Jenkins issue?

marc-hanheide, 2015-03-23:
it has changed: https://github.com/strands-project/strands_executive/commit/c9113ab7f3b5976313dd0fc086241d35be664fff#diff-4df48e1e34b2737c8f2760f44fd0fd45 and that's what causes it.

hawesie, 2015-03-23:
That was ages ago! I'll look.

hawesie, 2015-03-23:
Ah, but this is the first time it's been released since those changes I guess.

marc-hanheide, 2015-03-23:
yep

PR #140 (2015-03-23): Added   `strands_executive_msgs` to build depend list of wait_action. 
This should fix #138.
commit 740520b9 (2015-03-23): Added   `strands_executive_msgs` to build depend list of wait_action. This should fix #138.
closed 2015-03-23 by hawesie