strands-project/strands_executive #144: policy executor fail when current_node is none?

hawesie, 2015-03-27:
On the robot I just saw mdp policy execution fail repeatedly when /current_node was reporting none. I failed to capture the output though. Is this likely to be a problem?

Jailander, 2015-03-27:
never seen it shouldn't happen. please post output if seen again

bfalacerda, 2015-03-27:
was it during execution of a policy, or when one was received?

hawesie, 2015-03-27:
received i think

Jailander, 2015-03-27:
but how could it happen anyway? https://github.com/strands-project/strands_navigation/blob/indigo-devel/topological_navigation/scripts/execute_policy_server.py#L216

Jailander, 2015-03-27:
unless monitored navigation / move_base failed

bfalacerda, 2015-03-27:
it's probably on my stuff @Jailander

Jailander, 2015-03-27:
Ok, let me know for sure I'm doing tests before releasing the new fixes

hawesie, 2015-03-27:
Found the output

```
[INFO] [WallTime: 1427469813.276666] Concurrent state 'MONITORING' returned outcome 'succeeded' on termination.           │········
[INFO] [WallTime: 1427469813.277720] Preempt requested on action 'mdp_plan_exec/execute_policy'                           │········
[INFO] [WallTime: 1427469813.278318] Preempt on action 'mdp_plan_exec/execute_policy' cancelling goal:                    │········
task_type: 0                                                                                                              │········
target_id: WayPoint8                                                                                                      │········
[INFO] [WallTime: 1427469813.708602] Topological navigation execute policy action server exited with status: PREEMPTED    │········
[INFO] [WallTime: 1427469813.711192] Concurrent state 'MONITORED' returned outcome 'preempted' on termination.            │········
[INFO] [WallTime: 1427469813.743273] Concurrent Outcomes: {'MONITORED': 'preempted', 'MONITORING': 'succeeded'}           │········
[INFO] [WallTime: 1427469813.743722] Nav terminated with outcome aborted                                                  │········
[INFO] [WallTime: 1427469813.748777] State machine transitioning 'TASK_NAVIGATION':'aborted'-->'TASK_FAILED'              │········
[WARN] [WallTime: 1427469813.751182] Task failed                                                                          │········
[INFO] [WallTime: 1427469813.757813] Execution of task 9 failed                                                           │········
[INFO] [WallTime: 1427469813.758258] State machine terminating 'TASK_FAILED':'aborted':'aborted'                          │········
[INFO] [WallTime: 1427469909.514532] The specification is: R{"time"}min=? [ (F "‘WayPoint12’") ]                          │········
received: plan                                                                                                            │········
Error: Syntax error ("‘", line 1, column 22)                                                                              │········
Exception in thread "main" java.lang.NullPointerException                                                                 │········
        at prism.PrismPythonTalker.main(PrismPythonTalker.java:190)                                                       │········
[INFO] [WallTime: 1427469909.542682] Expected time from current node:                                                     │········
[ERROR] [WallTime: 1427469909.543215] Exception in your execute callback: could not convert string to float:              │········
Traceback (most recent call last):                                                                                        │········
  File "/opt/ros/indigo/lib/python2.7/dist-packages/actionlib/simple_action_server.py", line 299, in executeLoop          │········
    self.execute_callback(goal)                                                                                           │········
  File "/home/strands/1f_ws/src/strands_executive/mdp_plan_exec/scripts/mdp_policy_executor.py", line 134, in execute_poli│········
cy_cb                                                                                                                     │········
    self.generate_prod_mdp(specification)                                                                                 │········
  File "/home/strands/1f_ws/src/strands_executive/mdp_plan_exec/scripts/mdp_policy_executor.py", line 107, in generate_pro│········
d_mdp                                                                                                                     │········
    expected_time=float(self.prism_policy_generator.get_policy(specification))                                            │········
ValueError: could not convert string to float:                              
```

bfalacerda, 2015-03-28:
have you seen this again? Looking at this 

```
[INFO] [WallTime: 1427469909.514532] The specification is: R{"time"}min=? [ (F "‘WayPoint12’") ]                         
```

It looks like the ``execute_policy``action server received "'WayPoint12'", as its target, hence the extra single quotes there that screw it up

hawesie, 2015-03-30:
Not seen again.

bfalacerda, 2015-04-01:
closing this then

closed 2015-04-01 by bfalacerda