strands-project/strands_executive #146: priorities to guide "dropping" of tasks

marc-hanheide, 2015-03-30:
as discussed with @mudrole1 and @bfalacerda we should have support for priorities in tasks, both send to the scheduler and in the on-demand tasks. I'll quote from the gitter chats:

*bfalacerda March 28 2015*

i don't know exactly how long it wil take. bit if priority means "this tasks should not be dropped by the scheduler" and not "we should try to do this task as fast as possible", it shouldnt be to hard too add to the current version of the scheduler

*marc-hanheide March 28 2015*

Yes, I think the former will do the job for us... Our use case is as follows:

*bfalacerda March 28 2015*

it's just adding something to drop tasks when the current schedule doesnt fit in the time window, and use the priorities to decide which tasks should be dropped
the eay the scueduling problems are solved can be exactly the same

*marc-hanheide March 28 2015*

we have info_terminal almost continuously suggesting jobs, but as soon we have something of higher priority coming in, we want to give preference to that
we see it like this:
* info_terminal tasks: prio 1 (lowest)
* bellbot tasks: prio 2
* walking_group: prio 3
* charging: prio 4
* emergency: prio 100 (highest)

So, ideally, a currently running task will even be pre-empted, when a higher priority task is requested, and can't fit the current schedule. This would also apply to on-demand tasks
It's not about changing the order of scheduled tasks, but to kick tasks out if otherwise higher prio tasks would not be fitted in
The emergency task could also be implemented bypassing the scheduler, if that is needed.

*bfalacerda March 28 2015*

from what lenka told me, the main issue with that is lack of feedback
the scheduler drops tasks and the node that asked for it isnt informed

*hawesie March 28 2015*
  
on demand tasks bypass the scheduler anyway

*bfalacerda March 28 2015*
  
unless its publishing to a topic or something

*marc-hanheide March 28 2015*
  
more general:
* we have a "fremen"-routine submitting info_terminal tasks
* another gcal_routine submitting scheduled tasks defined in a google calendar
* and then there are on-demand tasks requested

I don't mind the lack of feedback

*bfalacerda March 28 2015*  

the other issue i see is the overloading of the scheduler, i'm not sure how well it will cope with so many tasks being given all the time
how frequent are the fremen tasks?

*marc-hanheide March 28 2015*

yes, on-demand tasks are not scheduled, but they are currently not featuring priorities either. So when we are doing something low prio, we should pre-empt it, and we should pre-empt high prio tasks when low-prio on demands are submitted every ten minutes currently so, only three at a time
(assuming a 30 minutes pre-scheduling window)
so, I don't think we would have more than 5-6 tasks at any given time produced by our routines

*bfalacerda March 28 2015*

ok, that might have to be tweaked but we need to get the dropping of low priority tasks first. let's wait for @mudrole1 to say something about this dropping of tasks using the priorities when it's the schedule doesnt fit

mudrole1, 2015-04-01:
@hawesie 
I dont have my notes from Vienna right now and I struggle to run test for task_executor. I run:
-roslaunch topological_utils dummy_topological_navigation.launch
-rosrun task_executor scheduled_task_executor.py but this complains that I need to define argument topological_map... and I cant find what is the name of the dummy one. What is it please?
-then I want to run my modified example_multi_add_client.py

hawesie, 2015-04-02:
Try  `roslaunch task_executor task-scheduler-top.launch` instead of the usual launch for task executor. Sorry, will update docs.

mudrole1, 2015-04-06:
Ooops, I am not sure, how did I make it, but I pushed my stuff directly to the strands-executive, and not to my branch :-/ I noticed it just now after three days of pushing stuff. Sorry, if you were confused :-/

mudrole1, 2015-04-06:
However, I did couple of things:
a) added "priority" field to task.msg
b) if all tasks cannot be executed, executor sort them based on priorities and try to throw away tasks with the lowest prio or 20% of them (whatever is smaller number)
c) if there is currently executed task and the new tasks are added and cannot be scheduled, the last task (others throw away) will be propagated as on demand task if the new task has higher priority, thus current task is preempted 
d) fixed bug in scheduler, which was causing that schedule was not found even there should be one
e) found and fixed two other bugs in scheduler - impossible tasks where scheduled
f) extended scheduler_node on some feedback, what is the case, if there is no schedule

I tested eveything on some limited amount of data. I need to perform more extensive testing this week. 

However, as I was almost finished with this, Ive realised that the functionality, which relates to point c:

Now, I am throwing tasks away, if they cannot be scheduled, and only for the last one, I check if the priority is higher than currently executed. The better (and correct) behaviour will be to try throw away those with lower priority than current task, but as task with higher prio is reached, then cause preemption. Thus. not throwing tasks with  higher prio than current task have.

I will work on this tomorrow.

hawesie, 2015-04-07:
@mudrole1 I have created a branch call `priorities` which contains all your changes. Please push there for updates then create a PR to `hydro-release` when you're ready.

hawesie, 2015-04-07:
Or even better, push everything to your fork and then PR as normal.

mudrole1, 2015-04-07:
Yep, I was thinking that I am pushing to my fork, however I wasnt :-/

mudrole1, 2015-04-07:
@hawesie 
What is the difference between self.execution_schedule.get_current_task() and self.active_task? Are they same (as the code looks like)?

hawesie, 2015-04-07:
In what file/class?

mudrole1, 2015-04-07:
sorry, task_executor/scheduled_task_executor.py uses self.execution_schedule.get_current_task(), but base_executor.py use active_task

hawesie, 2015-04-07:
`self.execution_schedule.get_current_task()` indicates which task is currently considered as active in the schedule. The base_executor `active_task` is the task that is actually being executed by the executor. These don't have to be the same. Usually the former is used to define the latter, but in the demand case it is done differently.

mudrole1, 2015-04-09:
@marc-hanheide  This should be solved by pull request #169

closed 2019-09-25 by bfalacerda