strands-project/strands_executive #151: wait action fails

bfalacerda, 2015-04-01:
```
[INFO] [WallTime: 1427898715.389491] Action wait_action started for task 2
[INFO] [WallTime: 1427898715.410265] target wait time: 2015-04-01 15:32:25
[ERROR] [WallTime: 1427898715.411067] Exception in your execute callback: global name 'WaitFeedback' is not defined
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/actionlib/simple_action_server.py", line 299, in executeLoop
    self.execute_callback(goal)
  File "/home/strands/1f_ws/src/strands_executive/wait_action/scripts/wait_node.py", line 49, in execute
    feedback = WaitFeedback()
NameError: global name 'WaitFeedback' is not defined
```

```
 [INFO] [WallTime: 1427898760.400250] Action wait_action started for task 3
[ERROR] [WallTime: 1427898760.500989] Exception in your execute callback: service [/wait_action/end_wait] already registe$
ed
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/actionlib/simple_action_server.py", line 299, in executeLoop
    self.execute_callback(goal)
  File "/home/strands/1f_ws/src/strands_executive/wait_action/scripts/wait_node.py", line 30, in execute
    Empty, self.end_wait)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 712, in __init__
    get_service_manager().register(self.resolved_name, self)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/service.py", line 116, in register
    raise ServiceException(err)
ServiceException: service [/wait_action/end_wait] already registered
```

@marc-hanheide it looks like this was introduced by your edits to the wait action

marc-hanheide, 2015-04-01:
Yep... I'll accept that  responsibility

marc-hanheide, 2015-04-01:
should be fixed in #153

commit 0d6e268f (2015-04-01): Merge pull request #153 from strands-project/patch-151

fixes #151
commit 8e8b1e24 (2015-04-01): fixes #151
closed 2015-04-01 by marc-hanheide