strands-project/strands_executive #152: use full travel times to waypoint vector

bfalacerda, 2015-04-01:
The  GetExpectedTravelTimesToWaypoint is being called redundantly  because the vector it returns is not being used at the scheduler level.

The scheduler should use it better

hawesie, 2015-04-02:
At my end I could consider epochs equivalent if they fall within a certain duration of each other. That would prevent me calling the travel time service for tasks that have execution times that are only seconds apart. Should I do that, or will you handle this as part of #154? My preference is that you handle it as then the caller can be ignorant of epoch window sizes.

bfalacerda, 2015-04-02:
yes, i'll do it, i also think it makes more sense at the mdp level

commit 3f2dcc35 (2015-04-09): Using full vector from mdp travel service.

This closes #152
closed 2015-04-09 by hawesie