strands-project/strands_executive #157: demand task: 'NoneType' object has no attribute 'close'

bfalacerda, 2015-04-06:
I see this happening once in a while in the routine window:

```
[INFO] [WallTime: 1428330625.937214] Idle for too long, going to charging point
[INFO] [WallTime: 1428330665.942632] Idle for too long, going to charging point
[ERROR] [WallTime: 1428330666.034750] bad callback: <bound method RobotRoutine._check_idle of <routine_behaviours.robot_routine.RobotRoutine object at 0x7f57c34187d0>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 702, in _invoke_callback
    cb(msg)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/routine_behaviours/robot_routine.py", line 195, in _check_idle
    self.on_idle()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/routine_behaviours/robot_routine.py", line 338, in on_idle
    self.demand_charge(self.idle_charge_duration)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/routine_behaviours/robot_routine.py", line 294, in demand_charge
    self.demand_task(charge_task)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/task_executor/demand_task] responded with an error: error processing request: 'NoneType' object has no attribute 'close'

[INFO] [WallTime: 1428330705.952216] Idle for too long, going to charging point
[INFO] [WallTime: 1428330745.958532] Idle for too long, going to charging point
```

bfalacerda, 2015-04-06:
Also this, but then I don't see any errors in the executive window:

```
[ERROR] [WallTime: 1428331585.856353] bad callback: <bound method RobotRoutine._check_idle of <routine_behaviours.robot_routine.RobotRoutine object at 0x7f57c34187d0>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 702, in _invoke_callback
    cb(msg)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/routine_behaviours/robot_routine.py", line 195, in _check_idle
    self.on_idle()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/routine_behaviours/robot_routine.py", line 338, in on_idle
    self.demand_charge(self.idle_charge_duration)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/routine_behaviours/robot_routine.py", line 294, in demand_charge
    self.demand_task(charge_task)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/task_executor/demand_task] responded with an error: service cannot process request: service [/mdp_plan_exec/get_expected_travel_times_to_waypoint] returned no response
```

bfalacerda, 2015-04-06:
this errors are with https://github.com/strands-project/strands_executive/pull/156 merged

bfalacerda, 2015-04-06:
It also looks like that after these errors on demanding the task, the tasks that were already scheduled are dropped from the schedule

hawesie, 2015-04-07:
Is this with @mudrole1 recent changes or without?

bfalacerda, 2015-04-07:
Without, I pulled only from your fork

hawesie, 2015-04-07:
It looks like there's some problem with the task being sent from `demand_charge` method in `robot_routine.py`. Can you print out the task there?

bfalacerda, 2015-04-07:
```
CHARGE TASK task_id: 0
start_node_id: ChargingPoint
end_node_id: ChargingPoint
action: wait_action
start_after: 
  secs: 0
  nsecs: 0
end_before: 
  secs: 0
  nsecs: 0
max_duration: 
  secs: 60
  nsecs: 0
execution_time: 
  secs: 0
  nsecs: 0
arguments: 
  - 
    first: "____time____"
    second: 0.0
  - 
    first: "____duration____"
    second: 60.0
[ERROR] [WallTime: 1428407797.405752] bad callback: <bound method RobotRoutine._check_idle of <routine_behaviours.robot_routine.RobotRoutine object at 
0x7fce06960890>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 702, in _invoke_callback
    cb(msg)
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 195, in _check_idle
    self.on_idle()
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 339, in on_idle
    self.demand_charge(self.idle_charge_duration)
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 295, in demand_charge
    self.demand_task(charge_task)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/task_executor/demand_task] responded with an error: error processing request: 'NoneType' object has no attribute 'close'
```

bfalacerda, 2015-04-07:
```
[INFO] [WallTime: 1428408121.799009] Idle for too long, going to charging point
CHARGE TASK task_id: 0
start_node_id: ChargingPoint
end_node_id: ChargingPoint
action: wait_action
start_after: 
  secs: 0
  nsecs: 0
end_before: 
  secs: 0
  nsecs: 0
max_duration: 
  secs: 60
  nsecs: 0
execution_time: 
  secs: 0
  nsecs: 0
arguments: 
  - 
    first: "____time____"
    second: 0.0
  - 
    first: "____duration____"
    second: 60.0
[ERROR] [WallTime: 1428408121.892189] bad callback: <bound method RobotRoutine._check_idle of <routine_behaviours.robot_routine.RobotRoutine object at 
0x7fce06960890>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 702, in _invoke_callback
    cb(msg)
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 195, in _check_idle
    self.on_idle()
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 339, in on_idle
    self.demand_charge(self.idle_charge_duration)
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 295, in demand_charge
    self.demand_task(charge_task)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/task_executor/demand_task] responded with an error: service cannot process request: service [/mdp_plan_exec/get_expected_tr
avel_times_to_waypoint] returned no response
```

bfalacerda, 2015-04-07:
the first error happens more often than the second. I wonder if both of them are in the same service call and there's some special case where the ``/mdp_plan_exec/get_expected_travel_times_to_waypoint`` service is doing something unexpected

bfalacerda, 2015-04-07:
just found an error printed in the executive indow:

```
[ERROR] [WallTime: 1428410710.881108] Error processing request: 'NoneType' object has no attribute 'close'
['Traceback (most recent call last):\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_reque
st\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/strands/1f_ws/src/strands_executive/task_e
xecutor/src/task_executor/base_executor.py", line 287, in demand_task_ros_srv\n    self.task_demanded(req.task, self.active_task)\n', '  File "/home/st
rands/1f_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 163, in task_demanded\n    success, added = self.try_schedule
(previously_scheduled)\n', '  File "/home/strands/1f_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 375, in try_sched
ule\n    if self.call_scheduler(to_schedule, lower_bound, current_id):\n', '  File "/home/strands/1f_ws/src/strands_executive/task_executor/scripts/sch
eduled_task_executor.py", line 262, in call_scheduler\n    resp = self.schedule_srv(tasks, earliest_start, current_id, self.get_duration_matrix(tasks))
\n', '  File "/home/strands/1f_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 184, in get_duration_matrix\n    return
 self.get_duration_matrix_mdp(tasks)\n', '  File "/home/strands/1f_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 217
, in get_duration_matrix_mdp\n    resp = self.get_mdp_vector(target, epoch)\n', '  File "/home/strands/1f_ws/src/strands_executive/task_executor/src/ta
sk_executor/base_executor.py", line 160, in get_mdp_vector\n    return self.expected_time_srv(target_waypoint=target, epoch=epoch)\n', '  File "/opt/ro
s/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__\n    return self.call(*args, **kwds)\n', '  File "/opt/ros/in
digo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 528, in call\n    transport.close()\n', '  File "/opt/ros/indigo/lib/python2.7/dis
t-packages/rospy/impl/tcpros_base.py", line 807, in close\n    self.socket.close()\n', "AttributeError: 'NoneType' object has no attribute 'close'\n"]
```

hawesie, 2015-04-07:
I think I have it. This is my bad. From https://github.com/strands-project/strands_executive/blob/hydro-release/task_executor/src/task_executor/base_executor.py#L158 if `task.start_after` is `None` then I suspect this will result.

bfalacerda, 2015-04-07:
There are many instances of 

``start_after: 
  secs: 0
  nsecs: 0``

without the error occuring though. Anyway, I'll merge your fix and check if it happens again

mudrole1, 2015-04-07:
I wonder why there are such instances.... Also, the previous tasks, has start_after 0, but also end_before zero... but not zero duration! Such a task should be invalid :-/

bfalacerda, 2015-04-07:
Still happens :(

Found another thing in the executive window:

```
[WARN] [WallTime: 1428416181.826516] Caught exception when getting expected time: transport error completing service call: unable to receive data from sender, check sender's logs for details
```

hawesie, 2015-04-07:
@mudrole1 demanded tasks typically won't have a window, but should have a duration, but nothing checks this AFAIK.

hawesie, 2015-04-07:
@bfalacerda  looking at the below output, please can you print out the values of `target` and `epoch` in base executor?
```
File "/home/strands/1f_ws/src/strands_executive/task_executor/src/ta
sk_executor/base_executor.py", line 160, in get_mdp_vector\n    return self.expected_time_srv(target_waypoint=target, epoch=epoch)\n'
```

bfalacerda, 2015-04-07:
```
[WARN] [WallTime: 1428417136.110595] lower bound 1428417121214463949
[INFO] [WallTime: 1428417136.111460] Including current task 19 in scheduling problem
TARGET WayPoint3
EPOCH 1428417900000000000
ENTERD SERVICE
ENTERED SET MDP
CALING PREDICT SERVICE
[INFO] [WallTime: 1428417136.141271] Next task to execute: 19
[INFO] [WallTime: 1428417136.142442] Execution of task 19 was requested
TARGET ChargingPoint
EPOCH 1428417136144623994
CALLED PREDICT SERVICE
LEAVING SET MDP
received: get_vector
[INFO] [WallTime: 1428417136.261298] Expected times to target state: [50.60424316934579, 118.4062987315817, 149.0150940851771, 71.46901411500292, 88.0698044559342, 109.8361135089342, 119.82333726493421, 19.471118294789893, 101.024$
3520361843, 40.41216097115532, 128.97733623593422, 140.33323684826848, 0.0, 61.510684773345794, 40.100853735345794, 28.86009337099842, 21.254479533, 27.688569498, 15.021295012, 24.756710623]
LEFT SERVICE
[WARN] [WallTime: 1428417136.262154] Caught exception when getting expected time: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] [WallTime: 1428417136.275895] State machine starting in initial state 'TASK_INITIALISATION' with userdata: 
        ['task']
TARGET ChargingPoint
EPOCH 1428417136282221078
ENTERD SERVICE
ENTERED SET MDP
CALING PREDICT SERVICE
CALLED PREDICT SERVICE
LEAVING SET MDP
received: get_vector
[INFO] [WallTime: 1428417136.438971] Expected times to target state: [53.51485158515357, 194.72006717018584, 151.92573106403222, 74.37962253080916, 90.98041287174105, 112.74672192474105, 122.73394568074106, 107.0637432513382, 184.$
7356823566928, 128.0047859276934, 131.88794465174107, 143.24384662402196, 87.59262495653994, 0.0, 43.01146215115357, 61.91401308272467, 56.64661221642393, 56.287847007423935, 67.81865335395602, 58.55676948642393]
LEFT SERVICE
[INFO] [WallTime: 1428417136.440204] expected_nav_duration:  10.0
```

bfalacerda, 2015-04-07:
maybe it's a concurrency issue again? there is a call to the get_expected_travel_times service made by the task executor, while it is also receiving calls from the scheduler.

It seems to only calculate one  expected times vector when that happens

hawesie, 2015-04-07:
Ah, great spot. The changes I made in this PR added a second path through my code to call the expect travel time service. Previously the single path had a lock on it, the second path doesn't have this lock. It will take me a bit longer to fix it, but it's easy to do.

bfalacerda, 2015-04-07:
shouldn't the rospy service stuff be able to handle that on its own?

bfalacerda, 2015-04-07:
or is it from the same service proxy?

hawesie, 2015-04-07:
It’s from the same service proxy.

hawesie, 2015-04-08:
I'll try and fix this today.

closed 2015-04-09 by bfalacerda