strands-project/strands_executive #158: Waiting for task_executor services causes top_nav called with Waypoint1

mudrole1, 2015-04-07:
@hawesie I have one for you too:

when I run first `rosrun task_executor example_lenka.py` (which is modification of example_add_multi.py, but shouldn't make diifrence here, I am just adding tasks in waypoint h_2).
It says "Waiting for task_executor service...". 

Then, I start `roslaunch task_executor task-scheduler-top.launch`. As it should start executing first task, it gives:
```[WARN] [WallTime: 1428421662.089216] Caught exception when getting expected time: service [/topological_navigation/travel_time_estimator] responded with an error: error processing request: Unknown start node: WayPoint1
```
and `roslaunch topological_utils dummy_topological_navigation.launch` says:

```
[ERROR] [WallTime: 1428421662.088778] Error processing request: Unknown start node: WayPoint1
['Traceback (most recent call last):\n', '  File "/opt/ros/hydro/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/lenka/phd/code/strands_ws/src/strands_navigation/topological_navigation/scripts/travel_time_estimator.py", line 36, in estimate_travel_time\n    raise Exception(\'Unknown start node: %s\' % req.start)\n', 'Exception: Unknown start node: WayPoint1\n']
```
All works fine, when I start first task-scheduler-top.launch and then example_lenka

hawesie, 2015-04-07:
Are you sure you’re not adding a task containing `WayPoint1` somewhere?

mudrole1, 2015-04-07:
I guess it relates to the fact that base_executor.py / __init__method sets current and closest nodes to Waypoint1, but that is not in dummy top_map... and I dont know what is happening beyond there

mudrole1, 2015-04-07:
My code is: 

#!/usr/bin/env python
import rospy
from strands_executive_msgs import task_utils
from strands_executive_msgs.msg import Task
from strands_executive_msgs.srv import AddTasks, SetExecutionStatus
from strands_navigation_msgs.msg import *
import sys

def get_services():
  # get services necessary to do the jon 
  add_tasks_srv_name = '/task_executor/add_tasks'
  set_exe_stat_srv_name = '/task_executor/set_execution_status'
  rospy.loginfo("Waiting for task_executor service...")
  rospy.wait_for_service(add_tasks_srv_name)
  rospy.wait_for_service(set_exe_stat_srv_name)
  rospy.loginfo("Done")
  add_tasks_srv = rospy.ServiceProxy(add_tasks_srv_name, AddTasks)
  set_execution_status = rospy.ServiceProxy(set_exe_stat_srv_name, SetExecutionStatus)
  return add_tasks_srv, set_execution_status

def create_wait_task(node, start, end, secs=rospy.Duration(5), prio=1):
  
  wait_task = Task(action='wait_action',start_node_id=node, end_node_id=node, max_duration = secs, start_after = start, end_before = end, priority = prio) #2280 passing time for 3 task, 2270 failing
  task_utils.add_time_argument(wait_task, rospy.Time()) # Lenka note: what is this???
  task_utils.add_duration_argument(wait_task, secs)
  return wait_task

if __name__ == '__main__':
  rospy.init_node("example_lenka")
  # get services to call into execution framework
  add_task, set_execution_status = get_services()

  #for simplicity all tasks happens in same locations
  
    
  #Test A - tasks with same priority
  #Travel duration is 10!
  #Test A1I - all feasible, arriving at the batch
  tasks = []
  st = rospy.Time.now()  
  for i in range(0,5):
    tasks.append(create_wait_task('h_2', st, st+rospy.Duration(65), rospy.Duration(5),2))
  task_id = add_task(tasks)
  set_execution_status(True) # Set the task executor running (if it isn't already)
  
  rospy.sleep(190)

hawesie, 2015-04-07:
> I guess it relates to the fact that base_executor.py / __init__method sets current and closest nodes to Waypoint1, but that is not in dummy top_map... and I dont know what is happening beyond there

That could be it. If you send a task before it has received an update from the topological localisation I would expect that to happen.

mudrole1, 2015-04-07:
Ok, and how to fix it? Should the service be published after the executor received the update from top localisation? Not before?

hawesie, 2015-04-07:
Yeah, that would be a good solution. 

This is a low priority thing IMO as most use cases will have topological localisation running before tasks are sent.

mudrole1, 2015-04-07:
yep, I spotted it accidentally... Will try do it later.

closed 2019-07-05 by bfalacerda