strands-project/strands_executive #163: field success is not a bool

bfalacerda, 2015-04-08:
```
[ERROR] [WallTime: 1428514962.118516] bad callback: <bound method RobotRoutine._check_idle of <routine_behaviours.robot_routine.RobotRoutine object at 0x7fcb5545c510>>
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 702, in _invoke_callback
    cb(msg)
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 195, in _check_idle
    self.on_idle()
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 339, in on_idle
    self.demand_charge(self.idle_charge_duration)
  File "/home/strands/1f_ws/src/strands_executive_behaviours/routine_behaviours/src/routine_behaviours/robot_routine.py", line 295, in demand_charge
    self.demand_task(charge_task)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/task_executor/demand_task] responded with an error: error processing request: field success is not a bool
```

bfalacerda, 2015-04-08:
this started happening  in bob sometime during the afternoon and now it's happening many times in a row

hawesie, 2015-04-08:
Are there errors on the executive window too?

bfalacerda, 2015-04-08:
yes, pretty much the same:

```

bfalacerda, 2015-04-09:
more complete output. Executive:

```

commit cfb8ced8 (2015-04-09): Correcting order of values returned from demand task service call.

Once the task_id number grew larger this was no longer interpreted (incorrectly) as a boolean, causing #163.

This fixes #163.
closed 2015-04-09 by hawesie