strands-project/strands_executive #165: Threads locks in scheduled_task_executor?

mudrole1, 2015-04-08:
Ive noticed the problem caused by two lines:

if(self.active_task is not None):
          if(low_prio.priority<= self.active_task.priority):
              #do something

If there is an active task when the first if is checked, but finishes just at that moment, second if throws exception. At least, I think that is the problem, as I tested the same code several time and exception occured couple of times and I noticed that the execution of task was just finishing when scheduler enter method containing the two aforementioned lines. I guess that I will need to add thread locks to ensure that exception will never happen, but this seems as more general problem as there are many similar places in scheduled_task_executor.py

I can try to fix this, but do you have any ideas than locks,  @hawesie , @bfalacerda ?

bfalacerda, 2015-04-08:
can you show the error and get the link to those lines in the code?

hawesie, 2015-04-09:
Yep, that will help us.

mudrole1, 2015-04-09:
Uaa, I shouldnt code in the evenings as my brain doesnt work :( There was another similar place in my code causing the issue and I fixed it easily. Just ignore this :D

closed 2015-04-09 by mudrole1