strands-project/strands_executive #166: Using task-scheduler-mdp causes the robot to go to wrong waypoints

gestom, 2015-04-08:
MDP task scheduler ends up in different nodes that is supposed to.

e.g.:

The task is (from /current_schedule topic):

    task_id: 4
    start_node_id: WayPoint4
    end_node_id: WayPoint4
    action: /info_task_server
   
But the robot goes to WayPoint7:
task_type: 0                                                                                                                                                                                                        │··························································
target_id: WayPoint4                                                                                                                                                                                                │··························································
[INFO] [WallTime: 1428526589.972562] Topological navigation execute policy action server exited with status: PREEMPTED                                                                                              │··························································
[INFO] [WallTime: 1428526589.977459] Concurrent state 'MONITORED' returned outcome 'preempted' on termination.                                                                                                      │··························································
[INFO] [WallTime: 1428526589.983733] Concurrent Outcomes: {'MONITORED': 'preempted', 'MONITORING': 'succeeded'}                                                                                                     │··························································
[INFO] [WallTime: 1428526589.984609] Nav terminated with outcome aborted                                                                                                                                            │··························································
[INFO] [WallTime: 1428526589.991230] State machine transitioning 'TASK_NAVIGATION':'aborted'-->'TASK_FAILED'                                                                                                        │··························································
[WARN] [WallTime: 1428526589.992366] Task failed                                                                                                                                                                    │··························································
[INFO] [WallTime: 1428526589.998124] start window: 1428526805.0                                                                                                                                                     │··························································
[INFO] [WallTime: 1428526589.999190]          now: 1428526589.997977972                                                                                                                                             │··························································
[INFO] [WallTime: 1428526589.999786] need to delay 215.2022028 for execution                                                                                                                                        │··························································

bfalacerda, 2015-04-08:
The navigation was preempted because it exceeded the predicted time for execution by more than whatever the threshold is.

gestom, 2015-04-09:
It did not looke like that - the robot actually stopped at waypoint7 and turned accordingly. It should not turn when the navigation is simple timeouted, no?

bfalacerda, 2015-04-09:
From what I see in the output you pasted, the navigation to the waypoint was preempted, and the robot *knows* he didn't reach is goal and will not try to execute anything there (Task failed). I don't know why he turned, the only thing I can say from the output is that the robot didn't go to a wrong waypoint, it simply had a navigation task preempted.

bfalacerda, 2015-04-09:
1428609605 converts to 4/9/2015, 9:00:05 PM in your timezone

closed 2019-07-05 by bfalacerda