strands-project/strands_executive #171: Check if the task is valid

mudrole1, 2015-04-09:
A task must be checked if it has valid time window. But what should be the procedure?

In my pull request, I have added to the scheduler that it checks, if the time window is valid. If not, the scheduler_node will refuse do anything... as result, executor will try to throw RANDOM task (here will be place to improve it to throw away the one causing problem).

 It might be better that executor check it and do something... throw or inform user or...?

hawesie, 2015-04-09:
The `add_task` service should check it, print a warning and drop it if there's a problem. I can also put a check into the scheduler node in the part where it translates between ROS and your representation, unless that's what you've already done.

mudrole1, 2015-04-09:
Yep, I did that in the scheduler_node. I can modify  the method add_tasks in scheduled_task_executor to check and drop.

hawesie, 2015-04-09:
I suggest dining it in `base_executor.py` in the add task and add tasks ros service methods. That way it's done for all future subclasses too.

mudrole1, 2015-04-09:
Cool, will do it

closed 2019-07-05 by bfalacerda