strands-project/strands_executive #173: Task execution ?

gestom, 2015-04-09:
rostopic echo /current_schedule reports:
-------------------------------------------------------------------------------------------------------------------------------
execution_queue: 
-
    task_id: 6
    start_node_id: WayPoint5
    end_node_id: WayPoint5
    action: /info_task_server
    start_after: 
      secs: 1428609605
      nsecs: 0
    end_before: 
      secs: 1428609898
      nsecs: 0
    max_duration: 
      secs: 180
      nsecs: 0
    execution_time: 
      secs: 1428609605
      nsecs: 0
    arguments: []
  - 
    task_id: 7
    start_node_id: WayPoint4
    end_node_id: WayPoint4
    action: /info_task_server
    start_after: 
      secs: 1428609905
      nsecs: 0
    end_before: 
      secs: 1428610198
      nsecs: 0
    max_duration: 
      secs: 180
      nsecs: 0
    execution_time: 
      secs: 1428609905
      nsecs: 0
    arguments: []
---
-------------------------------------------------------------------------------------------------------------------------------
And:
rosservice call /task_executor/get_execution_status 
status: True
-------------------------------------------------------------------------------------------------------------------------------

But the robot does not do anything.

bfalacerda, 2015-04-09:
use ```rosrun task_executor schedule_status.py ```, it makes it easier to check when the next task is going to start

bfalacerda, 2015-04-09:
1428609605 converts to 4/9/2015, 9:00:05 PM in your timezone

gestom, 2015-04-09:
yeap, that was the following task - but still nothing happened

gestom, 2015-04-09:
[INFO] [WallTime: 1428610330.958784] Waiting to execute /info_task_server @ ChargingPoint (task 5)
[INFO] [WallTime: 1428610330.959297] Execution to start at 2015-04-09 21:15:05
[INFO] [WallTime: 1428610330.960006] A further 3 tasks queued for execution

gestom, 2015-04-09:
let's see

gestom, 2015-04-09:
[INFO] [WallTime: 1428610826.754976] Waiting to execute /info_task_server @ WayPoint5 (task 7)
[INFO] [WallTime: 1428610826.755785] Execution to start at 2015-04-09 21:25:05
[INFO] [WallTime: 1428610826.756457] A further 3 tasks queued for execution
[INFO] [WallTime: 1428610831.754109] 
 
but nothing happens

hawesie, 2015-04-09:
Please paste the output of the task executive window for the duration this is happening (e.g. before and around the expected execution time) and also let us know which version of the software you are using (e.g. released package, current source version or some fork).

gestom, 2015-04-09:
Executive (fresh from github):
* remote origin
  Fetch URL: http://github.com/strands-project/strands_executive.git
  Push  URL: http://github.com/strands-project/strands_executive.git
  HEAD branch: hydro-release

Navigation (fresh from github):
* remote origin
  Fetch URL: http://github.com/strands-project/strands_navigation.git
  Push  URL: http://github.com/strands-project/strands_navigation.git
  HEAD branch: indigo-devel

Info_terminal  (fresh from github):
  Fetch URL: https://github.com/strands-project/aaf_deployment.git
  Push  URL: https://github.com/strands-project/aaf_deployment.git
  HEAD branch: indigo-devel
---------------------------------------------------------------------------------------
task_executor schedule_status.py:

[INFO] [WallTime: 1428613445.600358] Waiting to execute /info_task_server @ WayPoint4 (task 2)
[INFO] [WallTime: 1428613445.601344] Execution to start at 2015-04-09 22:05:05
[INFO] [WallTime: 1428613445.602265] A further 3 tasks queued for execution
---------------------------------------------------------------------------------------
Task execution: 

PARAMETERS                                                                                                                                       
 * /gcal_routine/calendar: henry.strands%40h...                                                                                                                             
 * /rosdistro: indigo                                                                                                                                                     
 * /rosversion: 1.11.10                                                                                                                                                   
 * /scheduled_task_executor/nav_service: mdp    
                                                                                                                                                               
core service [/rosout] found                                                                                                                                                      
[right-cortex-0]: launching nodes...                                                                                                                                                  
[right-cortex-0]: ROS_MASTER_URI=http://linda:11311                                                                                                 
[right-cortex-0]: process[frenap-1]: started with pid [19593]                                                                                                                                                   
[right-cortex-0]: process[special_waypoints_manager-2]: started with pid [19600]                                                                                                                                    
[right-cortex-0]: process[mdp_travel_time_estimator-3]: started with pid [19601]                                                                                                                                    
[right-cortex-0]: process[mdp_policy_executor-4]: started with pid [19604]                                                                                                                                          
[right-cortex-0]: process[schedule_server-5]: started with pid [19609]                                                                                                                                              
[right-cortex-0]: process[scheduled_task_executor-6]: started with pid [19618]                                                                                                                                      
[right-cortex-0]: process[wait_node-7]: started with pid [19623]                                                                                                                                                   
[right-cortex-0]: process[gcal_routine-8]: started with pid [19632]                                                                                                                                                 
[right-cortex-0]: ... done launching nodes 
----------------------------------------------------------------------------------------------------------------------
22:05 Nothing happens.
However, calling task_executor schedule_status.py returns:

[INFO] [WallTime: 1428613600.735127] Waiting to execute /info_task_server @ WayPoint4 (task 3)
[INFO] [WallTime: 1428613600.735559] Execution to start at 2015-04-09 22:10:05
[INFO] [WallTime: 1428613600.735917] A further 3 tasks queued for execution
[INFO] [WallTime: 1428613605.735262]

gestom, 2015-04-09:
It was working yesterday, the 'only' problem was that the robot went to wrong nodes

bfalacerda, 2015-04-09:
maybe instead of posting passive-agressive comments that have no purpose except annoy me, you can paste some output from the executive and the navigation as @hawesie asked, so we can look at what's going on?

bfalacerda, 2015-04-09:
it looks like frenap is being launched, which means you don't have the most recent version: https://github.com/strands-project/strands_executive/blob/hydro-release/task_executor/launch/task-scheduler-mdp.launch

That shouldnt be the problem though

gestom, 2015-04-09:
Sorry to annoy you by that comment. I just meant that there were only minor issues yesterday. I have pasted the output from the the task executive window as @hawesie asked, but the problem is that it did not print anything unusual.

bfalacerda, 2015-04-09:
```
right-cortex-0: process[frenap-1]: started with pid 19593: process[special_waypoints_manager-2]: started with pid 19600: process[mdp_travel_time_estimator-3]: started with pid 19601: process[mdp_policy_executor-4]: started with pid 19604: process[schedule_server-5]: started with pid 19609: process[scheduled_task_executor-6]: started with pid 19618: process[wait_node-7]: started with pid 19623: process[gcal_routine-8]: started with pid [19632]
```

Just this? if so its probably because you're launching it from the main machine to the right-cortex, so you dont see more output. For now it's better to either log in to right-cortex and launch there, or just launch everything in the main cpu so we can see something more being printed.

what about navigation?

closed 2015-06-03 by hawesie