strands-project/strands_executive #179: [prism] Error: Syntax error

marc-hanheide, 2015-04-14:
any suggestion what might be going wrong here @bfalacerda ?

```
[INFO] [WallTime: 1429028060.827619] The specification is R{"time"}min=? [ ( F "WayPoint33") ]                                                                │··············································································
received: get_vector                                                                                                                                          │··············································································
Error: Syntax error ("0", line 120, column 25)                                                                                                                │··············································································
Error: There is no currently loaded PRISM model to build                                                                                                      │··············································································
Exception in thread "main" java.lang.NullPointerException                                                                                                     │··············································································
        at prism.PrismPythonTalker.main(PrismPythonTalker.java:198)                                                                                           │··············································································
[WARN] [WallTime: 1429028060.882698] socket error while getting state vector                                                                                  │··············································································
[WARN] [WallTime: 1429028060.883341] socket error while getting state vector                                                                                  │··············································································
[ERROR] [WallTime: 1429028060.884332] Error processing request: [Errno 32] Broken pipe                                                                        │··············································································
['Traceback (most recent call last):\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n   │··············································································
 response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/opt/ros/indigo/lib/mdp_plan_exec/mdp_travel_time_estimator.py│··············································································
", line 35, in travel_times_to_waypoint_cb\n    state_vector=map(rospy.Duration, self.prism_estimator.get_state_vector(specification))\n', '  File "/opt/ros/i│··············································································
ndigo/lib/python2.7/dist-packages/mdp_plan_exec/prism_java_talker.py", line 64, in get_state_vector\n    self.sock.sendall("error\\n")\n', '  File "/usr/lib/p│··············································································
ython2.7/socket.py", line 224, in meth\n    return getattr(self._sock,name)(*args)\n', 'error: [Errno 32] Broken pipe\n']                                     │··············································································
[WARN] [WallTime: 1429028060.885404] Caught exception when getting expected time: service [/mdp_plan_exec/get_expected_travel_times_to_waypoint] responded wit│··············································································
h an error: error processing request: [Errno 32] Broken pipe                                                                                                  │··············································································
/opt/ros/indigo/lib/python2.7/dist-packages/smach_ros/introspection.py:132: SyntaxWarning: The publisher should be created with an explicit keyword argument '│··············································································
queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.                                               │··············································································
  data_class=SmachContainerStructure)                                                                                                                         │··············································································
/opt/ros/indigo/lib/python2.7/dist-packages/smach_ros/introspection.py:137: SyntaxWarning: The publisher should be created with an explicit keyword argument '│··············································································queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.                                               │··············································································  data_class=SmachContainerStatus)                                                                                                                            │··············································································[INFO] [WallTime: 1429028060.951343] State machine starting in initial state 'TASK_INITIALISATION' with userdata:                                             │··············································································        ['task']                                                                                                                                              │··············································································[INFO] [WallTime: 1429028061.113378] The specification is R{"time"}min=? [ ( F "WayPoint33") ]                                                                │··············································································[INFO] [WallTime: 1429028088.854113] updating from google calendar https://www.googleapis.com/calendar/v3/calendars/henry.strands%40hanheide.net/events?key=AI│··············································································zaSyC1rqV2yecWwV0eLgmoQH7m7PdLNX1p6a0&singleEvents=true&orderBy=startTime&maxResults=2500                                                                     │··············································································
[INFO] [WallTime: 1429028100.142806] Got a further 1 tasks to schedule                                                                                        │··············································································[

```

bfalacerda, 2015-04-14:
this means no mdp was loaded. Can you tell me when it occurred?

marc-hanheide, 2015-04-14:
We have it running on a side PC. Could that be a problem?

marc-hanheide, 2015-04-14:
We just ran standard task-execution-mdp.

marc-hanheide, 2015-04-14:
And it worked like that in Linda

bfalacerda, 2015-04-14:
so i assume this is happening all the time? 

the prism files are saved in ~/tmp/prism . Maybe Werner-s side pc doesnt have access somehow?

Can you check if there's stuff there? You should at least have a "top_map_name".mdp file (while the nodes are running). No other complaints on start up?

bfalacerda, 2015-04-14:
and ``prism-robots`` runs from ``/opt/prism-robots``

i start ``prism-robots`` subprocesses via ``popen``:

https://github.com/strands-project/strands_executive/blob/hydro-release/mdp_plan_exec/src/mdp_plan_exec/prism_java_talker.py#L14

marc-hanheide, 2015-04-14:
I checked all that. Prism starts but unfortunately the directory is deleted when prism is shutdown. So there was no such file. I think I tested while the nodes were running and the dir was empty but I'm not sure about that and I'm off-robot now.

bfalacerda, 2015-04-14:
ok, tell me more when you know. The only reason I see for the error is really a weird file access issue, either by the python node, or the java subprocess.

marc-hanheide, 2015-04-15:
This is the file:
```
strands@werner-right-cortex:~/tmp/prism$ ll time_estimator/aaf_predep.mdp 
-rw-rw-r-- 1 strands strands 36005 avril 15 09:47 time_estimator/aaf_predep.mdp
```

```
mdp
 
module M 
 
waypoint:[0..88] init 0;

[WayPoint62_Konferenzraum] (waypoint=0) -> 1.0:(waypoint'=86) + 0.0:(waypoint'=0);
[WayPoint62_WayPoint61] (waypoint=0) -> 1.0:(waypoint'=15) + 0.0:(waypoint'=0);
[WayPoint11_WayPoint20] (waypoint=1) -> 0.5:(waypoint'=17) + 0.5:(waypoint'=1);
[WayPoint11_WayPoint64] (waypoint=1) -> 1.0:(waypoint'=39) + 0.0:(waypoint'=1);
[WayPoint65_WayPoint13] (waypoint=2) -> 1.0:(waypoint'=74) + 0.0:(waypoint'=2);
[WayPoint65_WayPoint20] (waypoint=2) -> 0.5:(waypoint'=17) + 0.5:(waypoint'=2);
[WayPoint36_WayPoint27] (waypoint=3) -> 0.5:(waypoint'=21) + 0.5:(waypoint'=3);
[WayPoint36_WayPoint29] (waypoint=3) -> 0.01:(waypoint'=19) + 0.99:(waypoint'=3);
[ChargingPoint_Station] (waypoint=4) -> 0.5:(waypoint'=73) + 0.5:(waypoint'=4);
[WayPoint34_WayPoint48] (waypoint=8) -> 0.5:(waypoint'=83) + 0.5:(waypoint'=8);
[WayPoint34_WayPoint52] (waypoint=8) -> 0.5:(waypoint'=9) + 0.5:(waypoint'=8);
[WayPoint34_WayPoint75] (waypoint=8) -> 0.5:(waypoint'=10) + 0.5:(waypoint'=8);
[WayPoint52_WayPoint34] (waypoint=9) -> 0.5:(waypoint'=8) + 0.5:(waypoint'=9);
[WayPoint75_Lift1] (waypoint=10) -> 0.5:(waypoint'=30) + 0.5:(waypoint'=10);
[WayPoint75_WayPoint34] (waypoint=10) -> 0.5:(waypoint'=8) + 0.5:(waypoint'=10);
[WayPoint75_WayPoint48] (waypoint=10) -> 1.0:(waypoint'=83) + 0.0:(waypoint'=10);
[WayPoint76_WayPoint14] (waypoint=11) -> 0.5:(waypoint'=31) + 0.5:(waypoint'=11);
[WayPoint76_Lift1] (waypoint=11) -> 1.0:(waypoint'=30) + 0.0:(waypoint'=11);
[WayPoint76_WayPoint77] (waypoint=11) -> 0.5:(waypoint'=33) + 0.5:(waypoint'=11);
[WayPoint80_WayPoint64] (waypoint=12) -> 0.5:(waypoint'=39) + 0.5:(waypoint'=12);
[WayPoint80_Lift3] (waypoint=12) -> 0.5:(waypoint'=40) + 0.5:(waypoint'=12);
[WayPoint80_WayPoint4] (waypoint=12) -> 0.5:(waypoint'=72) + 0.5:(waypoint'=12);
[WayPoint53_WayPoint48] (waypoint=13) -> 0.5:(waypoint'=83) + 0.5:(waypoint'=13);
[WayPoint53_WayPoint82] (waypoint=13) -> 0.5:(waypoint'=54) + 0.5:(waypoint'=13);
[WayPoint56_WayPoint55] (waypoint=14) -> 0.5:(waypoint'=58) + 0.5:(waypoint'=14);
[WayPoint56_WayPoint83] (waypoint=14) -> 0.5:(waypoint'=60) + 0.5:(waypoint'=14);
[WayPoint61_WayPoint62] (waypoint=15) -> 1.0:(waypoint'=0) + 0.0:(waypoint'=15);
[WayPoint61_WayPoint84] (waypoint=15) -> 0.5:(waypoint'=63) + 0.5:(waypoint'=15);
[WayPoint20_WayPoint11] (waypoint=17) -> 1.0:(waypoint'=1) + 0.0:(waypoint'=17);
[WayPoint20_WayPoint65] (waypoint=17) -> 1.0:(waypoint'=2) + 0.0:(waypoint'=17);
[WayPoint21_WayPoint19] (waypoint=18) -> 1.0:(waypoint'=28) + 0.0:(waypoint'=18);
[WayPoint21_WayPoint28] (waypoint=18) -> 0.333333343267:(waypoint'=66) + 0.666666656733:(waypoint'=18);
[WayPoint29_WayPoint36] (waypoint=19) -> 0.5:(waypoint'=3) + 0.5:(waypoint'=19);
[WayPoint29_WayPoint38] (waypoint=19) -> 0.01:(waypoint'=24) + 0.99:(waypoint'=19);
[WayPoint25_WayPoint9] (waypoint=20) -> 0.5:(waypoint'=36) + 0.5:(waypoint'=20);
[WayPoint25_WayPoint16] (waypoint=20) -> 0.01:(waypoint'=22) + 0.99:(waypoint'=20);
[WayPoint27_WayPoint18] (waypoint=21) -> 0.5:(waypoint'=23) + 0.5:(waypoint'=21);
[WayPoint27_WayPoint36] (waypoint=21) -> 0.01:(waypoint'=3) + 0.99:(waypoint'=21);
[WayPoint16_WayPoint18] (waypoint=22) -> 0.01:(waypoint'=23) + 0.99:(waypoint'=22);
[WayPoint16_WayPoint25] (waypoint=22) -> 0.5:(waypoint'=20) + 0.5:(waypoint'=22);
[WayPoint18_WayPoint16] (waypoint=23) -> 0.5:(waypoint'=22) + 0.5:(waypoint'=23);
[WayPoint18_WayPoint27] (waypoint=23) -> 0.01:(waypoint'=21) + 0.99:(waypoint'=23);
[WayPoint38_WayPoint29] (waypoint=24) -> 0.5:(waypoint'=19) + 0.5:(waypoint'=24);
[WayPoint38_WayPoint40] (waypoint=24) -> 0.01:(waypoint'=25) + 0.99:(waypoint'=24);
[WayPoint40_WayPoint38] (waypoint=25) -> 0.5:(waypoint'=24) + 0.5:(waypoint'=25);
[WayPoint40_WayPoint44] (waypoint=25) -> 0.01:(waypoint'=84) + 0.99:(waypoint'=25);
[WayPoint42_WayPoint31] (waypoint=26) -> 1.0:(waypoint'=79) + 0.0:(waypoint'=26);
[WayPoint42_WayPoint33] (waypoint=26) -> 1.0:(waypoint'=41) + 0.0:(waypoint'=26);
[WayPoint46_WayPoint33] (waypoint=27) -> 0.5:(waypoint'=41) + 0.5:(waypoint'=27);
[WayPoint46_WayPoint44] (waypoint=27) -> 0.5:(waypoint'=84) + 0.5:(waypoint'=27);
[WayPoint19_WayPoint7] (waypoint=28) -> 0.01:(waypoint'=29) + 0.99:(waypoint'=28);
[WayPoint19_WayPoint21] (waypoint=28) -> 1.0:(waypoint'=18) + 0.0:(waypoint'=28);
[WayPoint7_Lift2] (waypoint=29) -> 1.0:(waypoint'=42) + 0.0:(waypoint'=29);
[WayPoint7_WayPoint12] (waypoint=29) -> 1.0:(waypoint'=43) + 0.0:(waypoint'=29);
[WayPoint7_WayPoint19] (waypoint=29) -> 0.666666686535:(waypoint'=28) + 0.333333313465:(waypoint'=29);
[WayPoint7_WayPoint66] (waypoint=29) -> 1.0:(waypoint'=85) + 0.0:(waypoint'=29);
[Lift1_WayPoint58] (waypoint=30) -> 0.5:(waypoint'=51) + 0.5:(waypoint'=30);
[... 498 lines cut ...]
   [WayPoint87_WayPoint15] (waypoint=82):10.382300345;
   [WayPoint87_WayPoint4] (waypoint=82):18.926730065;
   [WayPoint87_WayPoint8] (waypoint=82):10.251926065;
   [WayPoint87_WayPoint86] (waypoint=82):51.30308019;
   [WayPoint87_Lift3] (waypoint=82):33.888218924;
   [Wp87_WP2] (waypoint=82):32.739726087;
   [WayPoint48_WayPoint34] (waypoint=83):21.291766773;
   [WayPoint48_WayPoint53] (waypoint=83):9.604642017;
   [WayPoint48_WayPoint75] (waypoint=83):19.266015489;
   [WayPoint44_WayPoint40] (waypoint=84):22.622490771;
[... cut ...]
   [WayPoint66_WayPoint51] (waypoint=85):17.822292141;
   [Konferenzraum_WayPoint62] (waypoint=86):30.362159094;
   [Konferenzraum_WayPoint63] (waypoint=86):16.731895989;
   [Feuerloescher_WayPoint28] (waypoint=87):12.882593533;
   [Feuerloescher_WayPoint39] (waypoint=87):23.573732705;
   [Feuerloescher_WayPoint85] (waypoint=87):11.103685345;
   [WayPoint39_Feuerloescher] (waypoint=88):21.120124105;
   [WayPoint39_WayPoint63] (waypoint=88):12.104135607;
endrewards
```

Apologies for the long post...

marc-hanheide, 2015-04-15:
As I said, you can also go to https://lcas.lincoln.ac.uk/henry/ to login (window 9 in tmux)

marc-hanheide, 2015-04-15:
One observation: we have nodes that don't have any edges connected to them. Could that be a problem?

bfalacerda, 2015-04-15:
line 114 
[Cafeteria_WayPoint26_  0] (waypoint=45) -> 0.5:(waypoint'=81) + 0.5:(waypoint'=45);

Looks like there's a waypoiny called "WayPoint26_  0" . PRISM doesnt like the spaces there

bfalacerda, 2015-04-15:
sorry, not a waypoint, those are good. An edge_id called Cafeteria_WayPoint26_  0

bfalacerda, 2015-04-15:
is there more than one edge connecting these waypoints?

I also see an edge called Cafeteria_WayPoint26

marc-hanheide, 2015-04-15:
yes, that was it. Thanks! We have another issue now that @Jailander is looking into.

closed 2015-04-15 by marc-hanheide

bfalacerda, 2015-04-15:
ok. we need to discuss these disconnected waypoints to be sure there wont be any problems