strands-project/strands_executive #186: [task_executor] priorities order wrong?

marc-hanheide, 2015-04-17:
It seems to me that priority 2 tasks are dropped when priority 1 tasks are added. I was expecting it to be the other way around. Either I got it wrong or this is a bug? What is the priority order? @hawesie, do you know?

hawesie, 2015-04-17:
@mudrole1 handles priorites. I am ignorant here (and elsewhere!).

mudrole1, 2015-04-17:
the prority 2 is definitely  higher than 1, thus should be performed. However, it depends what task are you adding... are they on demand? are they interruptible? what are time windows and other properties? can you run it and copy here what executor launch file says?

marc-hanheide, 2015-04-17:
OK, I need to check this again. Can't do it now, I'll get back to you,

hawesie, 2015-06-03:
Is this fixed?

marc-hanheide, 2015-06-03:
This one is... there's still another issue with the priorities, but I still have no consistent picture of it. It seems that the scheduler does not "lock" anything while it is trying to fit the incoming tasks with different priorities, sometime resulting in intermediate schedules (with the wrong tasks removed) already becoming active. It seems to happen when adding a task and starting a new one happen at about the same time. But as I said, it would need systematic testing. Maybe one could provide a stress unit test for these cases...

closed 2015-06-03 by marc-hanheide