strands-project/strands_executive #203: [task_executor] regression, invalid assumption about WayPoint names

marc-hanheide, 2015-05-11:
The executor doesn't work for AAF anymore as there is a strange assumption hard-coded in there about the WayPoint names:
https://github.com/strands-project/strands_executive/blame/66b22e1fc432723738e92a3ef144759043bbfbdf/task_executor/src/task_executor/base_executor.py#L83

There is no such `WayPoint1` in the AAF map, and I think this is quite a bold assumption. As this is only updated when the robot is *at* a node (in its influence area) this will fail whenever it is away from one. Broke our neck today ;-)

maybe @hawesie or @bfalacerda can fix quick, please?

marc-hanheide, 2015-05-11:
Trace from gitter:

```
remote[werner-right-cortex-0]: [WARN] [WallTime: 1431337071.551334] Dropped 1 additional tasks which are no longer executable          │······································································
[WARN] [WallTime: 1431337076.625338] Caught exception when getting expected time: 'WayPoint1' is not in list                           │······································································
[WARN] [WallTime: 1431337085.504507] Caught exception when getting expected time: 'WayPoint1' is not in list                           │······································································
[WARN] [WallTime: 1431337198.492169] Caught exception when getting expected time: 'WayPoint1' is not in list                           │······································································
[WARN] [WallTime: 1431337200.353460] Caught exception when getting expected time: 'WayPoint1' is not in list                           │······································································
/opt/ros/indigo/lib/python2.7/dist-packages/smach_ros/introspection.py:132: SyntaxWarning: The publisher should be created with an expl│······································································
icit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information. │······································································
  data_class=SmachContainerStructure)                                                                                                  │······································································
/opt/ros/indigo/lib/python2.7/dist-packages/smach_ros/introspection.py:137: SyntaxWarning: The publisher should be created with an expl│······································································
icit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information. │······································································
  data_class=SmachContainerStatus)                                                                                                     │······································································
[WARN] [WallTime: 1431337202.183813] Caught exception when getting expected time: 'WayPoint1' is not in list                           │······································································
[ERROR] [WallTime: 1431337202.226890] Exception in your execute callback: None                                                         │······································································
Traceback (most recent call last):                                                                                                     │······································································
  File "/opt/ros/indigo/lib/python2.7/dist-packages/actionlib/simple_action_server.py", line 299, in executeLoop                       │······································································
    self.execute_callback(goal)                                                                                                        │······································································
  File "/home/strands/aaf_ws/src/strands_executive/mdp_plan_exec/scripts/mdp_policy_executor.py", line 140, in execute_policy_cb       │······································································
    self.generate_prod_mdp(specification)                                                                                              │······································································
  File "/home/strands/aaf_ws/src/strands_executive/mdp_plan_exec/scripts/mdp_policy_executor.py", line 111, in generate_prod_mdp       │······································································
    self.top_map_mdp.set_initial_state_from_waypoint(self.closest_waypoint)
```

there was an issue with the publisher (bloody can bus error on the robot), but this thing looks still suspicious, but we tested now and it worked... but it could lead to a racing condition, right?
but the real problem was the robot topic dying not this, so this has lower priority or can even be closed by @hawesie or @bfalacerda

hawesie, 2015-05-11:
It should be fixed, but in the future. Those values are only there until the first real values are received, which should be pretty soon after start-up. I did make a dumb assumption there though.

closed 2019-07-05 by bfalacerda