strands-project/strands_executive #206: Current schedule stops publishing

cdondrup, 2015-05-21:
Without any error or warning, the `/current_schedule` stops publishing. Observed it twice over the last two days.

Comment from @marc-hanheide: 
> It still seemed to work

hawesie, 2015-05-21:
The code that does the publishing is at the end of this method: https://github.com/strands-project/strands_executive/blob/hydro-release/task_executor/scripts/scheduled_task_executor.py#L661

If `try_schedule` is called and takes a long time then you won't see a schedule for a while. Is this a problem?

hawesie, 2015-05-21:
It would be relatively trivial to separate out the publishing into a separate thread.

cdondrup, 2015-05-21:
It seems to stop entirely at least for several hours.

marc-hanheide, 2015-05-26:
I'll close this as a duplicate of #208 as all evidence points the way this is the same problem.

marc-hanheide, 2016-04-11:
This still appears to be a problem, as seen again in AAF deployment. Nothing crashes, no log, but it just won't publish anymore. Sorry, no logs...

hawesie, 2016-04-11:
Reopen?

marc-hanheide, 2016-04-11:
re-open now, as we found the error in the logs:
```
[ERROR] [WallTime: 1460401217.577713] Exception in your execute callback: service [/topological_prediction/predict_edges] responded with an error: error processing request: 'NoneType' object has no attribute 'probabilities'
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/dist-packages/actionlib/simple_action_server.py", line 299, in executeLoop
    self.execute_callback(goal)
  File "/home/strands/aaf_ws/src/strands_executive/mdp_plan_exec/scripts/mdp_policy_executor.py", line 140, in execute_policy_cb
    self.generate_prod_mdp(specification)
  File "/home/strands/aaf_ws/src/strands_executive/mdp_plan_exec/scripts/mdp_policy_executor.py", line 112, in generate_prod_mdp
    self.top_map_mdp.set_mdp_action_durations(self.directory+self.file_name,rospy.Time.now())
  File "/home/strands/aaf_ws/src/strands_executive/mdp_plan_exec/src/mdp_plan_exec/top_map_mdp.py", line 102, in set_mdp_action_durations
    predictions=self.get_edge_estimates(epoch)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/topological_prediction/predict_edges] responded with an error: error processing request: 'NoneType' object has no attribute 'probabilities'

[INFO] [WallTime: 1460401217.579812] Concurrent state 'MONITORED' returned outcome 'aborted' on termination.
Exception in thread Thread-26:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 679, in schedule_tasks
    self.try_schedule(unscheduled)
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 499, in try_schedule
    sched_result = self.call_scheduler(to_schedule, lower_bound, current_id)
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 270, in call_scheduler
    resp = self.schedule_srv(tasks, earliest_start, current_id, self.get_duration_matrix(tasks)) # Lenka note: why does the service need earliest start and current id?
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 188, in get_duration_matrix
    return self.get_duration_matrix_mdp(tasks)
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 225, in get_duration_matrix_mdp
    resp = self.get_mdp_vector(target, epoch)
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/src/task_executor/base_executor.py", line 168, in get_mdp_vector
    return self.expected_time_srv(target_waypoint=target, epoch=epoch)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 515, in call
    responses = transport.receive_once()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 696, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 353, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 336, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
ServiceException: service [/mdp_plan_exec/get_expected_travel_times_to_waypoint] responded with an error: service cannot process request: service [/topological_prediction/predict_edges] responded with an error: error processing request: 'NoneType' object has no attribute 'probabilities'
```

this seems to have killed the scheduling thread

marc-hanheide, 2016-04-11:
So, the problem: when there is any exception in the schedule thread, it breaks forever... https://github.com/strands-project/strands_executive/blob/a74422a3f526867ae36667dc3f719a0aa6197af2/task_executor/scripts/scheduled_task_executor.py#L664-L689

Now, it would be interesting to hear from @bfalacerda what the above error could be?

So, 2 things needs to be done:

1. the above bug must be fixed
2. the schedule thread needs a catch statement that makes it more robust...

hawesie, 2016-04-11:
@marc-hanheide noted 2. For the live system, please can you stick a try/catch in there and restart? That will be quicker than getting the code updated and released. I can do that later though.

@bfalacerda the rest is you ;)

bfalacerda, 2016-04-11:
passing it do @Jailander as it seems to start with a ``/topological_prediction/predict_edges`` error. I'll also edit my code to catch the exception.

bfalacerda, 2016-04-11:
There should be some error messages in the nav tab from the above

marc-hanheide, 2016-04-11:
@bfalacerda please go to https://lcas.lincoln.ac.uk/henry/ and inspect yourself... My internet connection is currently so bad, I can maintain it for long enough to log in

marc-hanheide, 2016-04-11:
spoke too early... got it:

```
[ERROR] [WallTime: 1460401217.552281] Error processing request: 'NoneType' object has no attribute 'probabilities'
['Traceback (most recent call last):\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/strands/aaf_ws/src/strands_navigation/topological_navigation/scripts/topological_prediction.py", line 217, in predict_edge_cb\n    return self.get_predict(req.epoch.secs)\n', '  File "/home/strands/aaf_ws/src/strands_navigation/topological_navigation/scripts/topological_prediction.py", line 137, in get_predict\n    prob = list(ps.probabilities)\n', "AttributeError: 'NoneType' object has no attribute 'probabilities'\n"]
[ERROR] [WallTime: 1460401217.561388] Error processing request: 'NoneType' object has no attribute 'probabilities'
['Traceback (most recent call last):\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/strands/aaf_ws/src/strands_navigation/topological_navigation/scripts/topological_prediction.py", line 217, in predict_edg
```

cdondrup, 2016-04-11:
Apparently: https://github.com/strands-project/strands_navigation/blob/indigo-devel/topological_navigation/scripts/topological_prediction.py#L133-L137 breaks, so the fremen client returned a None value...

hawesie, 2016-04-11:
What, no `FibonacciResult`?!

marc-hanheide, 2016-04-11:
time to ping @gestom ?

bfalacerda, 2016-04-11:
I wonder how long until we can blame MIRA

cdondrup, 2016-04-11:
No error from the fremen server though... Btw. the robot tried to get from `ChargingPoint` to `ChargingPoint`. Not sure if that is important...

gestom, 2016-04-11:
Well, it might be that you are asking fremenserver to give you a prediction for an empty set of edges. 
'None' is then the correct answer.

cdondrup, 2016-04-11:
So one way to mitigate this would be to check the success flag or the state of the goal because it is set to aborted (just tried feeding the server rubbish). this would have to be done in the topological predictionby @Jailander 
The main question is, why was the model empty...

marc-hanheide, 2016-04-11:
so, I guess my assignee for this is correct ;-)

hawesie, 2016-04-11:
The client can't return None though, or at least I don't think so. Do we see the log error message here?

http://docs.ros.org/indigo/api/actionlib/html/simple__action__client_8py_source.html#l00151

I have a slight suspicion that this is due to concurrent calls to @Jailander's predict_edge_cb somehow corrupting the fremen action client. ROS does not serialise concurrent calls to services, it is up to the implementer if concurrent calls could occur.

marc-hanheide, 2016-04-11:
I can't see that message anywhere

marc-hanheide, 2016-04-11:
at least not on the console...

marc-hanheide, 2016-04-11:
But @hawesie maybe right, as also without that error message None could be returned, e.g.: http://docs.ros.org/indigo/api/actionlib/html/action__client_8py_source.html#l00180

marc-hanheide, 2016-04-11:
so, clearly, this calls for a Lock to be acquired, I guess.

Jailander, 2016-04-11:
Hi just saw this, it looks weird, is there anyway we can see the output of the navigation part for this case? It looks like topological prediction could have failed somewhere or a ros service problem as @hawesie suggested, if its the first case having the topological prediction output could give us more insight into why it failed, it cant be a wrong set of edges as that service predicts only existing edges, but it could be sonething else.

Jailander, 2016-04-11:
I'll implement a lock anyway

marc-hanheide, 2016-04-12:
Position of the charger:
[image]

marc-hanheide, 2016-04-15:
any progress on this, yet? @Jailander I really think a Lock would simply do the trick.

Jailander, 2016-04-15:
Yes I'll make a PR with it along the morning

Jailander, 2016-04-17:
This [PR](https://github.com/strands-project/strands_navigation/pull/303) is being tested on the robot with the lock implemented

closed 2016-04-27 by hawesie