strands-project/strands_executive #207: Dropping tasks for no apparent reason

cdondrup, 2015-05-21:
I observed a very particular scheduler behaviour (happened twice yesterday):

1. The current task appears twice in the `/current_schedule` (exactly the same even the task id) as soon as it becomes active.
1. When the task finishes, the scheduler reports 
 ```
[WARN] [WallTime: 1432131595.903224] Dropped 2 existing tasks which are no longer executable
```
 and then the next task disapeard

Example:

task1 15:00 - 15:05
task2 15:05 - 15:10
task3 15:10 - 15:15

task1 starts and has a duplicate entry in current schedule:

```
task1 id 5
task1 id 5
task2 id 6
task3 id 7
```

after task1 finishes, above warning is shown and the current schedule shows:

```
task1 id 5
task3 id 7
```

task3 executes normally at the given time but the same thing happens all over again and a possible task4 would be dropped.

hawesie, 2015-05-21:
@mudrole1 please could you look into this? I'm swamped with STRANDS management reports.

hawesie, 2015-05-21:
@cdondrup are all these tasks added normally or are any demanded? were they added together or separately?

hawesie, 2015-05-21:
Does this coincide with #206?

cdondrup, 2015-05-21:
They are added sequentially and are normal tasks (not demanded) (priority 1 - all the same priority) and it doesn't coincide with #206.

mudrole1, 2015-05-21:
Will have a look!

hawesie, 2015-05-21:
Sequentially and separated in time or sequentially quickly?

cdondrup, 2015-05-21:
In this case they were added quickly after each other.

cdondrup, 2015-05-21:
But as I said it happened for the following tasks as well.

mudrole1, 2015-05-23:
This behaviour is really strange. I can see why it dropped two tasks - as it is testing if a task can be executed within its time window depending when the previous task has finished. Here, it makes sense that duplicated task 5 was cancelled, and probably task 6 was affected by long execution of task 5, thus it couldn’t be executed.

However, I cant see reason why the task is duplicated. @cdondrup Did message "trying to drop X task with priority Y" occur?

One idea is referring to the fact that " were added quickly"... I experienced strange behaviour of addTask service before, when I was adding tasks quickly. Now, I am using delay of 2s when adding tasks within my test methods. I tried this today again, play around with delay, but I was unable to reproduce the strange behaviour, what I experienced month ago. 

Maybe, adding some delay between adding task might help.

cdondrup, 2015-05-23:
@mudrole1 not that I can recall. Maybe @marc-hanheide can have look in the logs if there are any.

Adding a delay shouldn't be a problem. We can try that.

marc-hanheide, 2015-05-26:
may also be a duplicate of #208?

closed 2019-07-05 by bfalacerda