strands-project/strands_executive #208: scheduler did not return from AddTasks Service

marc-hanheide, 2015-05-21:
Here is the relevant bit:

My routine called the AddTasks service at `13:40:04`:

Log of routine:
```
[rosout][INFO] 2015-05-21 13:40:04,957: setting priority of tasks to 2
[rosout][INFO] 2015-05-21 13:40:04,958: Sending 1 tasks to the scheduler
```

The call never returned  as the log stops there.

Log of `scheduled_executor`:
```
[rosout][INFO] 2015-05-21 13:38:58,841:  travel time:  0:02:32.481993
[rosout][INFO] 2015-05-21 13:38:58,842: need to delay 215.387061297 for execution
[rosout][INFO] 2015-05-21 13:38:58,842: Execution of task 36 failed
[rosout][INFO] 2015-05-21 13:38:58,843: State machine terminating 'TASK_FAILED':'aborted':'aborted'
[rosout][INFO] 2015-05-21 13:39:55,607: Got a further 1 tasks to schedule
[rosout][INFO] 2015-05-21 13:40:04,624: start window: 13:45:05
[rosout][INFO] 2015-05-21 13:40:04,625:          now: 13:40:02.922908
[rosout][INFO] 2015-05-21 13:40:04,625:  travel time:  0:02:32.481993
[rosout][INFO] 2015-05-21 13:40:04,626: need to delay 149.595099150 for execution
[rosout][INFO] 2015-05-21 13:40:04,627: Added 1 tasks into the schedule to get total of 4
```

The next is the reaction to the service being called (see times):

```
[rosout][INFO] 2015-05-21 13:40:04,993: Got a further 1 tasks to schedule
[rosout][INFO] 2015-05-21 13:40:13,316: No schedule found
[rosout][INFO] 2015-05-21 13:40:13,318: Schedule not found, trying to discard 1 tasks with priority 1
[rosout][INFO] 2015-05-21 13:40:20,401: No schedule found
[rosout][INFO] 2015-05-21 13:40:20,402: Schedule not found, trying to discard 1 tasks with priority 1
[rosout][INFO] 2015-05-21 13:40:27,507: start window: 13:45:05
[rosout][INFO] 2015-05-21 13:40:27,507:          now: 13:40:25.881983
[rosout][INFO] 2015-05-21 13:40:27,509:  travel time:  0:02:32.481993
[rosout][INFO] 2015-05-21 13:40:27,510: need to delay 126.636024176 for execution
[rosout][INFO] 2015-05-21 13:40:27,510: Added 1 tasks into the schedule to get total of 3
[rosout][INFO] 2015-05-21 13:42:35,969: start window: 13:45:05
[rosout][INFO] 2015-05-21 13:42:35,969:          now: 13:42:34.231156
[rosout][INFO] 2015-05-21 13:42:35,970:  travel time:  0:02:32.481993
[rosout][INFO] 2015-05-21 13:42:35,971: start window is open
[rosout][INFO] 2015-05-21 13:42:35,974: Next task to execute: 37
[rosout][INFO] 2015-05-21 13:42:35,975: Execution of task 37 was requested
[rosout][INFO] 2015-05-21 13:42:37,791: State machine starting in initial state 'TASK_INITIALISATION' with userdata: 
        ['task']
[rosout][INFO] 2015-05-21 13:42:39,604: expected_nav_duration:  152.481992783
[rosout][INFO] 2015-05-21 13:42:39,604: State machine transitioning 'TASK_INITIALISATION':'succeeded'-->'TASK_NAVIGATION'
[rosout][INFO] 2015-05-21 13:42:39,605: Concurrence starting with userdata: 
        ['task']
[rosout][INFO] 2015-05-21 13:42:39,606: Nav started to WayPoint68 for task 37, action /info_task_server

```

That's all I have observed, I'm afraid.

marc-hanheide, 2015-05-21:
not sure it's for @mudrole1 could also be for @hawesie

Reported by @cdondrup

hawesie, 2015-05-21:
What is the actual problem being reported?

marc-hanheide, 2015-05-21:
"scheduler did not return from AddTasks Service" as the title says? I called said service and it did not return, I believe. At least that's the evidence I've got. This is co-located with the `/current_schedule` not being published anymore, I think.

hawesie, 2015-05-21:
Ok, so add_tasks service call did not return? It's just that these are all different bits of the system so I want to be clear.

Do you have the task events being printed for this time?

hawesie, 2015-05-21:
Or could you check the contents of message_store/task_events to see if the addition of the task was logged. 

The `add_tasks` method obtains a lock, logs the events, then adds them to a queue. If the task addition was logged then it rules out the first two. 

Alternatively, is it possible that there is some transient network issue between the calling node and the task_executor?

hawesie, 2015-05-21:
And please link the code that does the addition, just for completeness.

marc-hanheide, 2015-05-21:
we indeed have network issues at the minute... I'm dealing with them now, but this happened long ago when network still was fine.

marc-hanheide, 2015-05-21:
event trace from mongo:

```
> db.task_events.find({"_meta.inserted_at": {$gt: ISODate("2015-05-21T11:39:00.0Z")}}).limit(20)
{ "_id" : ObjectId("555dc40b6812447b9f2cb76f"), "task" : { "start_node_id" : "ChargingPoint", "max_duration" : { "secs" : 293, "nsecs" : 0 }, "task_id" : 40, "execution_time" : { "secs" : 0, "nsecs" : 0 }, "end_before" : { "secs" : 1432209898, "nsecs" : 0 }, "end_node_id" : "ChargingPoint", "priority" : 1, "start_after" : { "secs" : 1432209605, "nsecs" : 0 }, "arguments" : [ ], "action" : "/info_task_server" }, "_meta" : { "stored_type" : "strands_executive_msgs/TaskEvent", "inserted_by" : "/scheduled_task_executor", "stored_class" : "strands_executive_msgs.msg._TaskEvent.TaskEvent", "inserted_at" : ISODate("2015-05-21T11:39:55.590Z") }, "time" : { "secs" : 1432208395, "nsecs" : 586416959 }, "event" : 1, "description" : "" }
{ "_id" : ObjectId("555dc4146812447b9f2cb82b"), "task" : { "start_node_id" : "WalkingGruppeStart", "max_duration" : { "secs" : 3900, "nsecs" : 0 }, "task_id" : 41, "execution_time" : { "secs" : 0, "nsecs" : 0 }, "end_before" : { "secs" : 1432213200, "nsecs" : 0 }, "end_node_id" : "WalkingGruppeStart", "priority" : 2, "start_after" : { "secs" : 1432209300, "nsecs" : 0 }, "arguments" : [ ], "action" : "walking_group_slow" }, "_meta" : { "stored_type" : "strands_executive_msgs/TaskEvent", "inserted_by" : "/scheduled_task_executor", "stored_class" : "strands_executive_msgs.msg._TaskEvent.TaskEvent", "inserted_at" : ISODate("2015-05-21T11:40:04.964Z") }, "time" : { "secs" : 1432208404, "nsecs" : 961337089 }, "event" : 1, "description" : "" }
{ "_id" : ObjectId("555dc4ad6812447b9f2cc3c7"), "task" : { "start_node_id" : "WayPoint68", "max_duration" : { "secs" : 293, "nsecs" : 0 }, "task_id" : 37, "execution_time" : { "secs" : 1432208705, "nsecs" : 0 }, "end_before" : { "secs" : 1432208998, "nsecs" : 0 }, "end_node_id" : "WayPoint68", "priority" : 1, "start_after" : { "secs" : 1432208705, "nsecs" : 0 }, "arguments" : [ ], "action" : "/info_task_server" }, "_meta" : { "stored_type" : "strands_executive_msgs/TaskEvent", "inserted_by" : "/scheduled_task_executor", "stored_class" : "strands_executive_msgs.msg._TaskEvent.TaskEvent", "inserted_at" : ISODate("2015-05-21T11:42:37.797Z") }, "time" : { "secs" : 1432208557, "nsecs" : 793101072 }, "event" : 3, "description" : "" }
{ "_id" : ObjectId("555dc4af6812447b9f2cc3eb"), "task" : { "start_node_id" : "WayPoint68", "max_duration" : { "secs" : 293, "nsecs" : 0 }, "task_id" : 37, "execution_time" : { "secs" : 1432208705, "nsecs" : 0 }, "end_before" : { "secs" : 1432208998, "nsecs" : 0 }, "end_node_id" : "WayPoint68", "priority" : 1, "start_after" : { "secs" : 1432208705, "nsecs" : 0 }, "arguments" : [ ], "action" : "/info_task_server" }, "_meta" : { "stored_type" : "strands_executive_msgs/TaskEvent", "inserted_by" : "/scheduled_task_executor", "stored_class" : "strands_executive_msgs.msg._TaskEvent.TaskEvent", "inserted_at" : ISODate("2015-05-21T11:42:39.611Z") }, "time" : { "secs" : 1432208559, "nsecs" : 607193946 }, "event" : 4, "description" : "" }
{ "_id" : ObjectId("555dc5256812447b9f2cc6e6"), "task" : { "start_node_id" : "WayPoint68", "max_duration" : { "secs" : 293, "nsecs" : 0 }, "task_id" : 37, "execution_time" : { "secs" : 1432208705, "nsecs" : 0 }, "end_before" : { "secs" : 1432208998, "nsecs" : 0 }, "end_node_id" : "WayPoint68", "priority" : 1, "start_after" : { "secs" : 1432208705, "nsecs" : 0 }, "arguments" : [ ], "action" : "/info_task_server" }, "_meta" : { "stored_type" : "strands_executive_msgs/TaskEvent", "inserted_by" : "/scheduled_task_executor", "stored_class" : "strands_executive_msgs.msg._TaskEvent.TaskEvent", "inserted_at" : ISODate("2015-05-21T11:44:37.723Z") }, "time" : { "secs" : 1432208677, "nsecs" : 718106985 }, "event" : 5, "description" : "" }
{ "_id" : ObjectId("555dc5256812447b9f2cc6e7"), "task" : { "start_node_id" : "WayPoint68", "max_duration" : { "secs" : 293, "nsecs" : 0 }, "task_id" : 37, "execution_time" : { "secs" : 1432208705, "nsecs" : 0 }, "end_before" : { "secs" : 1432208998, "nsecs" : 0 }, "end_node_id" : "WayPoint68", "priority" : 1, "start_after" : { "secs" : 1432208705, "nsecs" : 0 }, "arguments" : [ ], "action" : "/info_task_server" }, "_meta" : { "stored_type" : "strands_executive_msgs/TaskEvent", "inserted_by" : "/scheduled_task_executor", "stored_class" : "strands_executive_msgs.msg._TaskEvent.TaskEvent", "inserted_at" : ISODate("2015-05-21T11:44:37.734Z") }, "time" : { "secs" : 1432208677, "nsecs" : 729089975 }, "event" : 8, "description" : "" }
{ "_id" : ObjectId("555dc5376812447b9f2cc740"), "task" : { "start_node_id" : "Frisoer", "max_duration" : { "secs" : 293, "nsecs" : 0 }, "task_id" : 42, "execution_time" : { "secs" : 0, "nsecs" : 0 }, "end_before" : { "secs" : 1432210198, "nsecs" : 0 }, "end_node_id" : "Frisoer", "priority" : 1, "start_after" : { "secs" : 1432209905, "nsecs" : 0 }, "arguments" : [ ], "action" : "/info_task_server" }, "_meta" : { "stored_type" : "strands_executive_msgs/TaskEvent", "inserted_by" : "/scheduled_task_executor", "stored_class" : "strands_executive_msgs.msg._TaskEvent.TaskEvent", "inserted_at" : ISODate("2015-05-21T11:44:55.701Z") }, "time" : { "secs" : 1432208695, "nsecs" : 696650981 }, "event" : 1, "description" : "" }
{ "_id" : ObjectId("555dc6636812447b9f2ccb58"), "task" : { "start_node_id" : "WayPoint8", "max_duration" : { "secs" : 293, "nsecs" : 0 }, "task_id" : 43, "execution_time" : { "secs" : 0, "nsecs" : 0 }, "end_before" :  [... cut ...] }, "arguments" : [ ], "action" : "/info_task_server" }, "_meta" : { "stored_type" : "strands_executive_msgs/TaskEvent", "inserted_by" : "/scheduled_task_executor", "stored_class" : "strands_executive_msgs.msg._TaskEvent.TaskEvent", "inserted_at" : ISODate("2015-05-21T11:55:01.286Z") }, "time" : { "secs" : 1432209301, "nsecs" : 282519102 }, "event" : 17, "description" : "" }
```

I suppose the 2nd one corresponds to this addition. There were two additions successively with different priorities...

marc-hanheide, 2015-05-21:
call is here... https://github.com/strands-project/strands_executive/blob/hydro-release/gcal_routine/src/gcal_routine/queue_routine_runner.py#L83

you are right, my evidence isn't strong enough as the following logs are only debug and I didn't catch them :-( So, I *think* it didn't return as that thread ended there regardless what we tried, but I cannot be sure.

hawesie, 2015-05-21:
If that log is the addition and if 
```
[rosout][INFO] 2015-05-21 13:40:04,993: Got a further 1 tasks to schedule
```
indicates that the task that was added made it to the scheduler, then I am really at a loss to explain the blocking. If you look at the methods that handle the service call there is very little code after the log method has happened.

marc-hanheide, 2015-05-21:
OK... let's wait and see, but somehow we get the scheduler to hang up, not publishing the schedule and hanging somewhere... I keep my eyes open for it to happen again

hawesie, 2015-05-21:
Please keep the schedule_status and task_status scripts running as their output around that time (if available) will be useful.

marc-hanheide, 2015-05-21:
what are those scripts? We run everything from [this launch file](https://github.com/strands-project/aaf_deployment/blob/indigo-devel/aaf_bringup/launch/aaf_routine.launch). Don't see any scripts...

marc-hanheide, 2015-05-26:
OK, I have an update on this, as it seems to have happened again. No more publishing of `current_schedule`, no more task events indicating that tasks have been added, although other components return that task are being submitted.

Here some logs:

First, the event log as taken from mongo (with a projection to only display relevant fields):
```
> db.task_events.find({"_meta.inserted_at":{$gt: ISODate("2015-05-26T12:30:00.000Z")}},{"task.task_id":1, event: 1, "task.action": 1, "_meta.inserted_at":1})
{ "_id" : ObjectId("5564674a6812444738395d8d"), "task" : { "task_id" : 51, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:30:02.016Z") }, "event" : 3 }
{ "_id" : ObjectId("5564674d6812444738395d8e"), "task" : { "task_id" : 51, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:30:05.029Z") }, "event" : 4 }
{ "_id" : ObjectId("5564674f6812444738395d8f"), "task" : { "task_id" : 51, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:30:07.045Z") }, "event" : 5 }
{ "_id" : ObjectId("5564674f6812444738395d90"), "task" : { "task_id" : 51, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:30:07.052Z") }, "event" : 8 }
{ "_id" : ObjectId("5564686f6812444738395d9f"), "task" : { "task_id" : 56, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:34:55.065Z") }, "event" : 1 }
{ "_id" : ObjectId("556468936812444738395da5"), "task" : { "task_id" : 51, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:35:31.078Z") }, "event" : 10 }
{ "_id" : ObjectId("556468936812444738395da6"), "task" : { "task_id" : 51, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:35:31.086Z") }, "event" : 15 }
{ "_id" : ObjectId("5564699b6812444738395da8"), "task" : { "task_id" : 57, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:39:55.137Z") }, "event" : 1 }
{ "_id" : ObjectId("556469a26812444738395da9"), "task" : { "task_id" : 53, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:40:02.543Z") }, "event" : 3 }
{ "_id" : ObjectId("556469a56812444738395daa"), "task" : { "task_id" : 53, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:40:05.605Z") }, "event" : 4 }
{ "_id" : ObjectId("556469a76812444738395db1"), "task" : { "task_id" : 53, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:40:07.622Z") }, "event" : 5 }
{ "_id" : ObjectId("556469a76812444738395db2"), "task" : { "task_id" : 53, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:40:07.632Z") }, "event" : 8 }
{ "_id" : ObjectId("55646ac76812444738395db4"), "task" : { "task_id" : 58, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:44:55.200Z") }, "event" : 1 }
{ "_id" : ObjectId("55646acc6812444738395db5"), "task" : { "task_id" : 59, "action" : "/bellbot" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:45:00.348Z") }, "event" : 1 }
{ "_id" : ObjectId("55646aeb6812444738395dcb"), "task" : { "task_id" : 53, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:45:31.680Z") }, "event" : 10 }
{ "_id" : ObjectId("55646aeb6812444738395dcc"), "task" : { "task_id" : 53, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:45:31.690Z") }, "event" : 15 }
{ "_id" : ObjectId("55646b7b6812444738395dcd"), "task" : { "task_id" : 55, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:47:55.318Z") }, "event" : 3 }
{ "_id" : ObjectId("55646b7c6812444738395dce"), "task" : { "task_id" : 55, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:47:56.794Z") }, "event" : 4 }
{ "_id" : ObjectId("55646ba66812444738395dd4"), "task" : { "task_id" : 55, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:48:38.847Z") }, "event" : 6 }
{ "_id" : ObjectId("55646ba66812444738395dd5"), "task" : { "task_id" : 55, "action" : "/info_task_server" }, "_meta" : { "inserted_at" : ISODate("2015-05-26T12:48:38.854Z") }, "event" : 15 }
```

Obviously the last ask added was task `59`. It is now 13:47 UTC, so the last entry in the table above is about an hour old. In the mean time, other components submitted tasks:

e.g. infremen:
```
[ INFO] [1432644895.345243893]: Task 61 created at WayPoint8 for timeslot 159 on 2015-05-26_15:15:00, between 1432646105 and 1432646398. 
[ INFO] [1432645195.469283994]: Task 62 created at WayPoint12 for timeslot 160 on 2015-05-26_15:20:00, between 1432646405 and 1432646698. 
[ INFO] [1432645210.472053445]: Infremen: there were 2 interactions.
[ INFO] [1432645495.531910535]: Task 63 created at WayPoint26 for timeslot 161 on 2015-05-26_15:25:00, between 1432646705 and 1432646998. 
[ INFO] [1432645795.584257391]: Task 64 created at ChargingPoint for timeslot 162 on 2015-05-26_15:30:00, between 1432647005 and 1432647298. 
[ INFO] [1432646095.635473545]: Task 65 created at Frisoer for timeslot 163 on 2015-05-26_15:35:00, between 1432647305 and 1432647598. 
[ INFO] [1432646395.685578094]: Task 67 created at WayPoint68 for timeslot 164 on 2015-05-26_15:40:00, between 1432647605 and 1432647898. 
[... cut ...]
[rospy.internal][INFO] 2015-05-26 15:32:43,298: topic[/task_executor/events] adding connection to [/control_pc/rosbridge_websocket], count 1
[rospy.internal][INFO] 2015-05-26 15:33:34,687: topic[/current_schedule] adding connection to [/rostopic_5268_1432647214259], count 1
```

So, we see again, that no new tasks were added neither was the schedule published anymore. This only seems to happen whenever we have a higher priority tasking popping up (`bellbot` had priority 2 here!). And it doesn't always happen. Looks very much like a racing condition. Any idea what to do @hawesie @mudrole1? Anything else you want me to check?

marc-hanheide, 2015-05-26:
I have also just looked at the heartbeat of SMACH, and here's the readable output:

```
---
header: 
  seq: 38
  stamp: 
    secs: 1432648781
    nsecs: 357580900
  frame_id: ''
path: /TASK_ROOT/TASK_EXECUTION
initial_states: ['MONITORING', 'MONITORED']
active_states: []
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 39
  stamp: 
    secs: 1432648781
    nsecs: 356810092
  frame_id: ''
path: /TASK_ROOT/TASK_NAVIGATION
initial_states: ['MONITORING', 'MONITORED']
active_states: []
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 40
  stamp: 
    secs: 1432648783
    nsecs: 139426946
  frame_id: ''
path: /TASK_ROOT
initial_states: ['TASK_INITIALISATION']
active_states: ['None']
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 41
  stamp: 
    secs: 1432648783
    nsecs: 363188982
  frame_id: ''
path: /TASK_ROOT/TASK_NAVIGATION
initial_states: ['MONITORING', 'MONITORED']
active_states: []
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 42
  stamp: 
    secs: 1432648783
    nsecs: 363020896
  frame_id: ''
path: /TASK_ROOT/TASK_EXECUTION
initial_states: ['MONITORING', 'MONITORED']
active_states: []
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 43
  stamp: 
    secs: 1432648785
    nsecs: 144013881
  frame_id: ''
path: /TASK_ROOT
initial_states: ['TASK_INITIALISATION']
active_states: ['None']
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 44
  stamp: 
    secs: 1432648785
    nsecs: 367511034
  frame_id: ''
path: /TASK_ROOT/TASK_NAVIGATION
initial_states: ['MONITORING', 'MONITORED']
active_states: []
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 45
  stamp: 
    secs: 1432648785
    nsecs: 367975950
  frame_id: ''
path: /TASK_ROOT/TASK_EXECUTION
initial_states: ['MONITORING', 'MONITORED']
active_states: []
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 46
  stamp: 
    secs: 1432648787
    nsecs: 148088932
  frame_id: ''
path: /TASK_ROOT
initial_states: ['TASK_INITIALISATION']
active_states: ['None']
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---

```

I can't really read this matrix, but to me it looks like it's flipping between states?

marc-hanheide, 2015-05-26:
just discovered there was also this ERROR log:
```
[ERROR] [WallTime: 1432644518.227982] Error getting MDP next state. Aborting.
[WARN] [WallTime: 1432644518.850201] Task failed
Exception in thread Thread-27:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 679, in schedule_tasks
    self.try_schedule(unscheduled)
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 503, in try_schedule
    sub_additional, throwen_away, priority_reached = self.throw_away_tasks(to_schedule, 0.2) #throw away 20% if tasks have same prio
  File "/home/strands/aaf_ws/src/strands_executive/task_executor/scripts/scheduled_task_executor.py", line 402, in throw_away_tasks
    rospy.logingo('Schedule not found, trying to discard %s tasks with priority higher prio %s', str(throw_num), str(low_prio.priority))
AttributeError: 'module' object has no attribute 'logingo'
```

Obviously, should be loginfo, but maybe that killed it? This is actually well aligned in time with the other errors above, e.g. `[rosout][WARNING] 2015-05-26 14:48:38,850: Task failed`

marc-hanheide, 2015-05-26:
Just from completeness, here the SMACH topic when the scheduler is in a task working properly:
```
---
header: 
  seq: 44
  stamp: 
    secs: 1432650812
    nsecs: 257320880
  frame_id: ''
path: /TASK_ROOT/TASK_NAVIGATION
initial_states: ['MONITORING', 'MONITORED']
active_states: []
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 45
  stamp: 
    secs: 1432650812
    nsecs: 257240056
  frame_id: ''
path: /TASK_ROOT
initial_states: ['TASK_INITIALISATION']
active_states: ['TASK_EXECUTION']
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 46
  stamp: 
    secs: 1432650812
    nsecs: 258138895
  frame_id: ''
path: /TASK_ROOT/TASK_EXECUTION
initial_states: ['MONITORING', 'MONITORED']
active_states: ['MONITORED', 'MONITORING']
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 47
  stamp: 
    secs: 1432650814
    nsecs: 264117956
  frame_id: ''
path: /TASK_ROOT
initial_states: ['TASK_INITIALISATION']
active_states: ['TASK_EXECUTION']
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 48
  stamp: 
    secs: 1432650814
    nsecs: 265280008
  frame_id: ''
path: /TASK_ROOT/TASK_EXECUTION
initial_states: ['MONITORING', 'MONITORED']
active_states: ['MONITORED', 'MONITORING']
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
header: 
  seq: 49
  stamp: 
    secs: 1432650814
    nsecs: 263664007
  frame_id: ''
path: /TASK_ROOT/TASK_NAVIGATION
initial_states: ['MONITORING', 'MONITORED']
active_states: []
local_data: ?}qUtaskqcstrands_executive_msgs.msg._Task
--
---
```

marc-hanheide, 2015-05-26:
I submitted #209 to probably fix this?

hawesie, 2015-05-26:
Lenka needs to buy us all beer for that typo! I can't check stuff as away from computer on family holiday, but that could be an issue! Not sure why it locks up afterwards though.

marc-hanheide, 2015-05-26:
She can buy one for @kunzel as I owe him... :-)

mudrole1, 2015-05-27:
Oh ... :hankey: Sorry! But one more reason to come back to C++ :)

hawesie, 2015-06-03:
Can this be closed now?

marc-hanheide, 2015-06-03:
not until we had :beers:

hawesie, 2015-06-03:
@mudrole1  will need to bring :moneybag: to the review

marc-hanheide, 2015-06-03:
ok, I think she got the message.

closed 2015-06-03 by marc-hanheide