strands-project/strands_executive #217: Publish IDs of dropped and completed tasks

gestom, 2016-01-26:
It would be nice if a task ID was published whenever a task is completed.
Similarly, it would be nice if a task ID was published whenever a task is dropped.

(Needed for AAF deployment).

hawesie, 2016-01-26:
Yep! Do you have a rough deadline by when you'd like this?

gestom, 2016-01-26:
Monday would be nice (sorry for that).

hawesie, 2016-01-26:
Monday is difficult. I could probably guarantee it by the Friday after. It might be @mudrole1 could do it sooner?

gestom, 2016-01-26:
I can also try it myself, if you give me hint where it could be in the code.

hawesie, 2016-01-26:
I'm happy to try to get it in for Monday, but just not promise it.

The places to look are:

https://github.com/strands-project/strands_executive/blob/hydro-release/task_executor/scripts/scheduled_task_executor.py

and perhaps its superclass:

https://github.com/strands-project/strands_executive/blob/hydro-release/task_executor/src/task_executor/sm_base_executor.py

Also:

https://github.com/strands-project/strands_executive/blob/hydro-release/task_executor/src/task_executor/execution_schedule.py

mudrole1, 2016-01-26:
I am now finishing something else, I can update you about my free time Wed evening.

mudrole1, 2016-01-29:
The tasks are dropped in the file scheduled_task_executor.py in the method try_schedule line 426... Dropped tasks are kept in the list "dropped_tasks" so the tasks is to just publish this (or maybe just IDs)

About publishing the ID when a task is completed - Do you need the id after it is completed? I thought in our initial discussion, you need to get ID back when you create a new task...  
If you need completion, I need to have a deeper look where it is done

hawesie, 2016-02-03:
@mudrole1 did you do anything on this? Now the IJCAI papers are done I can look.

mudrole1, 2016-02-03:
Nope, sorry :-/ (as no one was urging I completely forgot about it )

hawesie, 2016-02-03:
Ok, I’ll do it.

gestom, 2016-02-03:
Hi, I just started to look into it, I think that I might have it done by Friday as part of planned infoterminal improvements.

hawesie, 2016-02-03:
I’m literally looking at the code now. Let me know if you want me to add the change or leave it to you.

gestom, 2016-02-03:
Go ahead then, thanks!

hawesie, 2016-02-03:
When you say 'completed' do you care whether they succeeded or failed?

hawesie, 2016-02-03:
Actually it doesn't matter as both are in there.

In the current system you can listen to the topic `/task_executor/events` which publishes `TaskEvent` messages. These messages have the task itself (including its id) and an event field. The event field is assigned based on the constants in the TaskEvent msg definition. To get an idea of what this gives you, run `rosrun task_executor task_status.py` and you'll get the events printed out as they occur.

hawesie, 2016-02-03:
The above does task completion, but not dropping. I've added a new branch to this repo [publish_dropped_tasks](https://github.com/strands-project/strands_executive/tree/publish_dropped_tasks) which adds my initial attempt at publishing on dropped tasks. This publishes `TaskEvent` messages on dropping, and there is also `/task_executor/dropped` topic which has just the dropped tasks published on it. This is actually redundant give the event topic, so I'll probably remove this again. 

This branch works for the cases I can easily test, but I'm not 100% confident it covers all the cases of task dropping. If you could run this branch for your use cases and let me know if it does what you want, that would be great.

hawesie, 2016-02-08:
@gestom please let me know how this works for you.

closed 2016-04-27 by hawesie