strands-project/strands_executive #251: Negative expected times!

hawesie, 2016-05-11:
Now on Betty:

```

For p = 0.897075355053, r = 0.897075355053 the minimum expected  cummulative cost to satisfy specification is -178.8197232141465

Exporting expected times until no more progression over product to file "/home/strands/tmp/prism/policy_executor_extended//guarantees3.vect"...

Value in the initial state: -178.8197232141465

Time for model checking: 0.042 seconds.

Result: -178.8197232141465 (value in the initial state)
[0]
[3]
[INFO] [WallTime: 1462948932.482769]  received feedback PENDING
[WARN] [WallTime: 1462948932.483036] Expected duration was less that 0, giving a default of 5 minutes
[INFO] [WallTime: 1462948933.238140] Another 359.25 seconds until expected policy completion
[INFO] [WallTime: 1462948943.239816] Another 349.24 seconds until expected policy completion
```

hawesie, 2016-05-11:
I'm assuming it's down to this in door prediction?

```
─────
errors: []
1462948932 ['door_WayPoint2_WayPoint1_time', 'door_WayPoint14_WayPoint15_time', 'door_WayPoint6_WayPoint7_time', 'door_WayPoint10_WayPoint11_time'] [0, 2, 4, 1]
success: 0
message: Predicted the probabilities of 4 states 
probabilities: [-1.0, -1.0, -1.0, -1.0]
entropies: []
errors: []
requesting prediction for time 1462948932
1462948932 ['door_WayPoint2_WayPoint1_res', 'door_WayPoint14_WayPoint15_res', 'door_WayPoint6_WayPoint7_res', 'door_WayPoint10_WayPoint11_res'] [0, 0, 0, 0]
success: 0
message: Predicted the probabilities of 4 states 
probabilities: [-1.0, -1.0, -1.0, -1.0]
entropies: []
errors: []
1462948932 ['door_WayPoint2_WayPoint1_time', 'door_WayPoint14_WayPoint15_time', 'door_WayPoint6_WayPoint7_time', 'door_WayPoint10_WayPoint11_time'] [0, 2, 4, 1]
success: 0
message: Predicted the probabilities of 4 states 
probabilities: [-1.0, 
```

bfalacerda, 2016-05-11:
Yeah, I forgot to pull https://github.com/strands-project/strands_apps/pull/51.

Just did it and restarted the node, reopen if it happens again

hawesie, 2016-05-11:
This is happening on Betty right now.

bfalacerda, 2016-05-11:
:( @Jailander I have the commit that added the lock there, looks like it's something else...

bfalacerda, 2016-05-11:
restarted it and it happened again

Jailander, 2016-05-11:
Oh no... Ok, I'll take more preventive measures and more debug messages maybe we find out what is going on

closed 2016-11-26 by bfalacerda