strands-project/strands_executive #253: empty waypoint makes guarantees service sad

hawesie, 2016-05-13:
```
[WARN] [WallTime: 1463144488.589195] Caught exception in the mdp_exec loop: service [/mdp_plan_exec/get_guarantees_for_co_safe_task] responded with an error: error processing request: ''
[INFO] [WallTime: 1463144490.592507] Checking for normal tasks to fit into available time: 1200.00 secs
(F executed_n5_wait_action_at_=1)
[ERROR] [WallTime: 1463144490.605583] Error processing request: ''
['Traceback (most recent call last):\n', '  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 623, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/strands/tsc_ws/src/strands_executive/mdp_plan_exec/scripts/mdp_task_guarantees_estimator.py", line 44, in get_guarantees_cb\n    self.current_extended_mdp.add_extra_domain(req.spec.vars, req.spec.actions)\n', '  File "/home/strands/tsc_ws/src/strands_executive/mdp_plan_exec/src/mdp_plan_exec/top_map_mdp.py", line 310, in add_extra_domain\n    self.transitions.append(self.action_msg_to_transition(action, waypoint, use_expected_duration))\n', '  File "/home/strands/tsc_ws/src/strands_executive/mdp_plan_exec/src/mdp_plan_exec/top_map_mdp.py", line 272, in action_msg_to_transition\n    pre_conds["waypoint"]=self.get_waypoint_var_val(waypoint)\n', '  File "/home/strands/tsc_ws/src/strands_executive/mdp_plan_exec/src/mdp_plan_exec/top_map_mdp.py", line 163, in get_waypoint_var_val\n    return self.props_def[waypoint_prop].conds[\'waypoint\']\n', "KeyError: ''\n"]
[WARN] [WallTime: 1463144490.605955] Caught exception in the mdp_exec loop: service [/mdp_plan_exec/get_guarantees_for_co_safe_task] responded with an error: error processing request: ''

```

bfalacerda, 2016-05-13:
yes it does. you want me to make this more robust, or were you expecting expected times for all possible starting waypoints?

hawesie, 2016-05-13:
This is my issue

hawesie, 2016-05-13:
I'll just assume a small default.

closed 2016-06-06 by hawesie