strands-project/strands_executive #258: bug in mdp executor when no action post-conditions are specified

hawesie, 2016-06-06:
I tried to specify an mdp action outcome with no post-conditions (i.e. it left the state-variable unchanged) but this caused something to make PRISM upset.

bfalacerda, 2016-06-06:
can you show me how you defined the outcome and what prism told you was upsetting it?

hawesie, 2016-06-06:
https://github.com/strands-project/planning_tutorial/blob/indigo-devel/scripts/sweep_at_waypoints.py

with line 53 commented out.

bfalacerda, 2016-06-07:
is it too bad requiring you to define that, even if you define it as the same state?

hawesie, 2016-06-07:
It’s not the end of the world, but it feels ugly. I suspect this pattern (positive outcome changes state, negative does not) will reoccur in future apps.

bfalacerda, 2016-06-07:
ok, it's not too hard to solve, I'll do it on Friday if that's alright

hawesie, 2016-06-07:
There’s no rush.

closed 2019-07-05 by bfalacerda