strands-project/strands_executive #281: executor should check travel time before opening window

hawesie, 2016-11-23:
currently travel time is assumed to happen within the window.

gestom, 2016-11-23:
So what will happen if I submit a task that has start_after = 0, max_duration = 400 and end_before = 500 and the expected nav time will be 200 ? Is the task dropped ?

hawesie, 2016-11-23:
It depends on the current time. If the current time is 0 then yes, as it's impossible to get to the location and do the task in the window. If the current time is -100 then it will not get dropped.

hawesie, 2016-11-23:
(this was the case in the previous scheduler implementation too)

gestom, 2016-11-24:
Would it be possible to have another task parameter called 'min_duration', which, if set to a nonzero value, would be used instead of max_duration for this window check?

PR #282 (2016-11-25): Dealing with windows more actively
This checks navigation time before opening a window. Also it adds option to terminate execution of a batch of task when the lowest time window has passed.

To use this option add `close_windows:=true` to the command of launching the system, e.g.

```
roslaunch --wait task_executor mdp-executor.launch close_windows:=true
```

This fixes #281 and #283
commit c666600d (2016-11-25): Using expected nav time before window opens.

This fixes #281
closed 2016-11-25 by hawesie