strands-project/strands_executive #321: Scheduler is working for the ROS Kinetic branch?

EliosSs, 2022-02-15:
Hey to all. I hope you are well.

I was looking at the strands executive project but I noticed that for kinetic branch the "scheduler" folder is missing. 
MILP scheduler is not implemented for kinetic? or maybe it was embedded in other node?

hawesie, 2022-02-16:
We replaced the MILP scheduler with an MDP-based one in the second version of the system: https://github.com/strands-project/strands_executive/blob/kinetic-devel/task_executor/scripts/mdp_task_executor.py

There are some tutorial examples for some of the planning parts of the system here: https://github.com/strands-project/planning_tutorial/tree/indigo-devel

hawesie, 2022-02-16:
That call is in a node we stopped using a long time ago, sorry.

EliosSs, 2022-02-16:
ok ok thanks very much Nick! I will check that information in order to know what is the call the scheduler now.

closed 2022-02-16 by hawesie

EliosSs, 2022-02-16:
Hey again Nick. I was checking the code and the tutorial you just send me. Actually I have more doubts about it. Can I ask here just with the closed issue? or maybe I need to open a new one?

Best regards and thank you a lot for your help.

EliosSs, 2022-02-16:
Well I think I can ask some of the doubts that I have over here. I can create another issue if those cannot be here.
1. (maybe a noobie question) You send me the link of the "mdp_task_executor.py" file from task executor package. But checking the file and the whole package I just realized that there is not any node registered in the CMakeLists.txt file. it is any reason for this? Maybe this is noobie question but I am trying to understand why? 

2. Actually I am running ROS melodic, and when I was checking the planing_tutorial repository (indigo-devel branch) that you sent me I realized that you do have the same packages for ROS melodic, which made me think that it could run in ROS melodic but when I was checking the tutorial for the exercise 1a I saw this line:
`roslaunch mdp_plan_exec mdp_plan_exec_extended.launch` 

and checking in the strands executive repo I noticed that this file does not exist for some of the branches, not even the kinetic branch but for indigo-devel branch. So maybe I am missing something regarding the latest repo? or do you have an updated repo for newer ROS versions like melodic or noetic?

Another think was that I got an error of "task_executor" missing package when I was following the planing_tutorial package, which lead me to my 3rd doubt.

3. In any of the examples I didn't see any command to run "task_executor" package. not even in the configuration part and it does not seem to be a dependency of any of the packages? does it have any reason?

Sorry if some of the doubts are stupid or too obvious. I really appreciate your help.

Thanks very much in advance.

hawesie, 2022-02-17:
1. `CMakeLists.txt` is used for building/installing things, and mostly on the C++ side. Since the node is Python it doesn't need to be built.
2. I'd suggest following the tutorial from the other link I provided. I think we ran that in melodic. 
3. If you don't follow the tutorial, then you could try this: https://github.com/strands-project/strands_executive/blob/kinetic-devel/README.md This mentions that package and steps you through some of its usage.

EliosSs, 2022-02-17:
Ehhh yeah as python files are just scripts but it is not necessary to list the .py files inside the CMake file in order to recognize the ROS nodes? ahmmmm I didn't know that, thanks. 

Ahmmm yeah I am trying to follow the planing_tutorial package that you sent to me. 
But when I run the catkin_make command I receive this error:

`-- +++ processing catkin package: 'planning_tutorial'
-- ==> add_subdirectory(planning_tutorial)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Could NOT find task_executor (missing: task_executor_DIR)
-- Could not find the required component 'task_executor'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "task_executor"
  with any of the following names:

    task_executorConfig.cmake
    task_executor-config.cmake

  Add the installation prefix of "task_executor" to CMAKE_PREFIX_PATH or set
  "task_executor_DIR" to a directory containing one of the above files.  If
  "task_executor" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  planning_tutorial/CMakeLists.txt:7 (find_package)`
  
  For that is my 3rd doubt. It should not be an automatic dependency? or maybe is something related just to ROS melodic?

hawesie, 2022-02-17:
Those packages are not in the main ROS repo so are not automatically provided. Therefore you need to clone the strands-executive repo into your catkin ws. You used to provide pre-built packages (some info here: https://github.com/strands-project/planning_tutorial/blob/master/doc/tutorial_prep.md), but I don't think we did that for melodic.

EliosSs, 2022-02-21:
oh ok ok Nick. Thanks very much for your answer. 
I have another question left. There is any documentation that I could learn to understand a little more about "ltl" and the conversions between "spec" and "mdp_tasks"?

hawesie, 2022-02-22:
I'd suggest starting with one of our videos, e.g. https://www.youtube.com/watch?v=axUmQNDcsT8 describing the work in https://www.robots.ox.ac.uk/~nickh/papers/LacerdaIJRR.pdf